MCUdude/MiniCore #188: How can I decide in a library, that MiniCore is used.

ArminJo, 2021-04-24:
Is it possible to have something like
```
name=Digistump AVR Boards
versionnum.major=1
versionnum.minor=7
versionnum.patch=3
versionnum.postfix=

version={versionnum.major}.{versionnum.minor}.{versionnum.patch}{versionnum.postfix}

build.versiondefines=-DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} -DDIGISTUMPCORE="{version}" -DDIGISTUMPCORE_MAJOR={versionnum.major}UL -DDIGISTUMPCORE_MINOR={versionnum.minor}UL -DDIGISTUMPCORE_PATCH={versionnum.patch}UL
...
recipe.c.o.pattern+ recipe.cpp.o.pattern+ recipe.S.o.pattern="...{build.versiondefines} {compiler.c.extra_flags}...

```
as can be seen in 
https://github.com/ArminJo/DigistumpArduino/blob/master/digistump-avr/platform.txt
and 
https://github.com/SpenceKonde/megaTinyCore/blob/9beeed82b144f5b1fbd5d8384bf9a319cbdc10b8/megaavr/platform.txt

I had issues with IRremote not able to detect your core.
Thanks
Armin

MCUdude, 2021-04-25:
How does this fix the IRRemote library issues? You'll most likely have to enable LTO when compiling the library.

ArminJo, 2021-04-26:
The library compiles fine without lto, only this warning pops always up.
But if I can detect the core, I can suppress this warning for the core, since it is mostly not necessary!

MCUdude, 2021-04-26:
Ah, you're the maintainer of IRremote! FYI there are some compatibility issues with my other cores. 

* Starting with [MegaCoreX](https://github.com/MCUdude/MegaCoreX), ATmega4809 is the only target that compiles.
* With [MegaCore](https://github.com/MCUdude/MegaCore) I'm not able to compile for ATmega2561, ATmega640, AT90CAN128, AT90CAN64 and AT90CAN32. For the rest of the targets I can't get it to compile without LTO. Here's the error. BTW I'm getting the same error with other cores too, including [MajorCore](https://github.com/MCUdude/MajorCore), [MightyCore](https://github.com/MCUdude/MightyCore) and MiniCore.
```
In file included from /Users/hans/Documents/Arduino/libraries/IRremote/src/IRremote.h:181:0,
                 from /Users/hans/Documents/Arduino/libraries/IRremote/examples/SendDemo/SendDemo.ino:43:
/Users/hans/Documents/Arduino/libraries/IRremote/src/IRSend.cpp.h: In member function 'void IRsend::begin(bool, uint8_t)':
/Users/hans/Documents/Arduino/libraries/IRremote/src/IRSend.cpp.h:80:15: error: call to 'UsageError' declared with attribute error: Version > 3.0.1
     UsageError(
     ~~~~~~~~~~^
             "Error: You must use begin(<sendPin>, <EnableLEDFeedback>, <LEDFeedbackPin>) if SEND_PWM_BY_TIMER is not defined or USE_NO_SEND_PWM is defined!");
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

On topic:
I'm not sure what warning you're referring to. Can you post it here? Also, is there a specific version of MiniCore that causes trouble? Would it help if `#define MINICORE` existed, so you could check if MiniCore is in use or not?

ArminJo, 2021-04-26:
> Error: You must use begin(<sendPin>, <EnableLEDFeedback>,

This is the warning I am referring to 😀 
And this is the reason I need these platform.txt extensions or at least a working check like `#if defied(MINICORE)`
For your tests, you can suppress the error by activating the line #define SUPPRESS_ERROR_MESSAGE_FOR_BEGIN in IRremote.h

> Would it help if #define MINICORE existed, so you could check if MiniCore is in use or not?

This is exact the thing I need at minimum!!! 👍 

> Starting with MegaCoreX, ATmega4809 is the only target that compiles.

If you get other errors, than the one above, please let me know. I am planning a hotfix release this week.

MCUdude, 2021-04-26:
>> Would it help if #define MINICORE existed, so you could check if MiniCore is in use or not?

> This is exact the thing I need at minimum!!! 👍

Excellent. The reason why I'm a bit hesitant about adding version numbers in the platform.txt file is that it won't be automatically added to the PlatformIO installation since PlatformIO doesn't use platform.txt, but rather its own thing. I should come up with something eventually, for now, I'm going to add the following macros for all my cores:
```cpp
#define MINICORE

#define MIGHTYCORE

#define MEGACORE

#define MAJORCORE

#define MEGACOREX

```

> If you get other errors, than the one above, please let me know. I am planning a hotfix release this week.

Thanks. As far as my testing goes, the only targets that don't work at all are:
 * ATmega808/809, ATmega1608/1609, ATmega3208/3209, ATmega4808 (MegaCoreX)
 * ATmega2561, AT90CAN128/64/32 (MegaCore)

ArminJo, 2021-04-26:
> Thanks. As far as my testing goes, the only targets that don't work at all are:
> 
>     * ATmega808/809, ATmega1608/1609, ATmega3208/3209, ATmega4808 (MegaCoreX)
> 
>     * ATmega2561, AT90CAN128/64/32 (MegaCore)

Maybe they are not supported in terms of initializing a timer for the 50 us interrupt.

ArminJo, 2021-04-26:
I included the check with https://github.com/Arduino-IRremote/Arduino-IRremote/commit/5dd88831f263b106acddc4df73409a9946096300
And [here](https://github.com/Arduino-IRremote/Arduino-IRremote/commit/9e9df236bcaa4fc949a5dc1c2c4b5ae47f30a690) the addition of the megaAVR 0-series

MCUdude, 2021-04-26:
> I included the check with Arduino-IRremote/Arduino-IRremote@5dd8883
And here the addition of the megaAVR 0-series

Thanks! Macros for my other cores are now available too.

ArminJo, 2021-04-26:
Which of your core does have lto disabled at default? All these I have to include in my check list.

MCUdude, 2021-04-26:
> Which of your core does have lto disabled at default? All these I have to include in my check list.

All, except for MegaCoreX.

ArminJo, 2021-04-26:
> All, except for MegaCoreX.

Done, but I amended this and created no additional commit.

closed 2021-08-15 by MCUdude