MCUdude/MiniCore #192: Correct times with non-standard clock speeds

Takuyax, 2021-06-04:
Hi! First thanks a lot for your great work.

I'm currently running an Atmega328p with a 6.4 MHz crystal, to get multiple 25 KHz PWM outputs for controlling PWM fans.
I export the binary and flash the 328p (without bootloader) using a minipro, and also use it to set the fuses. So I can run it at that speed despite the missing preset, and everything works great so far. The only problem is that I don't get correct times when using e.g. micros() or delayMicroseconds().
What would be the best way to correct the time? Can I simply make a new entry for 6.4 MHz in the boards.txt?

MCUdude, 2021-06-04:
Hi!

You can try to make a 6.4MHz option in the boards.txt file. I think the timing functions should be fairly accurate with arbitrary speeds like 6.4MHz, but I don't remember exactly. I'm sure @cburstedde knows, he is the brains behind the current timing implementation!

Takuyax, 2021-06-07:
Thank you! I gave it a try and it works without any problems so far.

MCUdude, 2021-06-07:
Perfect! For reference, here is the PR that made this possible (an accuracy of at least 60ppm with any F_CPU)

https://github.com/MCUdude/MCUdude_corefiles/pull/32

closed 2021-06-07 by MCUdude

cburstedde, 2021-06-07:
> Thank you! I gave it a try and it works without any problems so far.

Thanks for copying me and glad it works.  The only function without automatic
calculation is delayMicroseconds, for which you'll be using the 6 MHz version.