MCUdude/MiniCore #193: Shift+Upload DOES NOT remove bootloader

duraluminium, 2021-06-05:
I just wanted to clear up some behavior, as it was confusing me quite a bit. 

After burning bootloaders to some ATTINY chips using ATTinyCore, and getting used to the option of uploading a sketch to the ATTiny using Arduino as ISP just by clicking the upload button, the behavior in MINICORE was configured differently, namely that after burning a bootloader you have to select "Upload with programmer (Shift+Upload)" to upload a sketch to my ATmega 328 (just clicking the upload button with the 328 board selected causes an error, as the IDE tries to upload the sketch to the Arduino UNO and not the 328). After figuring out that I needed the "upload with programmed" option, I managed to upload my sketch, but the MINICORE readme specifies:

> Keep your programmer connected, and hold down the shift button while clicking Upload. This will erase the bootloader and upload your code using the programmer tool.

This lead me to believe that even after burning fuses AND bootloader (so I could upload using serial), uploading a sketch through ICSP would remove the bootloader I just burned and stop me from later uploading via serial. (The behavior in ATTinyCore leaves the bootloader alone after uploading a sketch using ICSP)

This turns out to be **FALSE**, as uploading a sketch 'with programmer' does not modify the bootloader (as long as the same board parameters are used as when burning the bootloader). 

So burn bootloader->upload sketch with programmer->upload sketch with serial works perfectly fine, contrary to the readme instructions saying the bootloader would be erased.
This is in line with the ATTinyCore behavior, which is nice, keeping things standardized. I just wanted to give people a heads up, and perhaps encourage an update to the wording in the readme, as this leads people (like me) to believe that after burning a bootloader, I have to switch to serial (rewire) to get a sketch onto the chip, when all along I could have uploaded that first sketch with the ICSP wiring the chip is already connected to.

Thanks for all the hard work MCUdude. Cheers

MCUdude, 2021-06-05:
Hi!

Thanks for pointing out the errors in the README.
It's true that the bootloader isn't removed when you shift+upload when a bootloader option is selected, and this is different from how the official Arduino AVR boards work. With MiniCore, you'll have to select "No bootloader" to wipe the bootloader, which IMO makes sense. I'll update the README soon.

closed 2021-06-06 by MCUdude

pouriap, 2022-02-08:
Well this explains why I have not been able to program my ATmega8 s with bootloader. The instructions in the readme are completely incorrect. 

I don't know how OP was even able to figure out that you have to use SHIFT+Upload to program using rx/tx as it is supposed to do the exact opposite.

Also, IMO this is a completely misleading behavior. It should be the same as it is written in the readme. i.e. clicking upload uploads using rx/tx and SHIFT+Upload uploads using programmer.

Why should "Upload using programmer" actually upload *NOT* using programmer? That makes no sense.

Also if both ICSP upload and rx/tx upload is done by pressing SHIFT+Upload, then how does the core even know which one to do? I am very confused still.

I had MiniCore installed a few years ago and programmed ATmega8 with it with rx/tx with no problems. This new behavior is extremely confusing and in contrast to Arduino IDE and also in contrast to how MiniCore used to work. 

Please change it back.

(It goes without saying that I have the greatest respect and appreciation for the author of this library, my life would suck big time without minicore and microcore. But I was very frustrated and confused as to why I can't get the bootloader to work.)

MCUdude, 2022-02-08:
Sorry, but I don't understand what you're talking about. Are you sure you don't have a corrupted installation?

I've inserted an ATmega8 into my UNO board, connected both the USB cable from the UNO to my computer, and the USBasp to the ISP pins.

* I'm able to but the bootloader by selecting USBasp as the programmer and clicking "Burn Bootloader".
* When I click the upload button, and "Bootloader: Yes (UART0)" is selected, it uploads like normal over UART.
* When doing shift + upload, it uses the USBasp programmer to upload regardless of the "Bootloader" option in the tools menu. Shift + upload will always upload using the programmer.
* When I compile with "Bootloader: Yes (UART)", the available flash size is 7680 bytes, for "Bootloader: None", 8192 bytes.

The upload button will always upload using the bootloader if "Bootloader: Yes (UART0)" is selected in the tools menu.

pouriap, 2022-02-08:
Hi. 
Thanks for the quick reply.

- I put the ATmega8 on breadboard with minimal circuit and a 16MHz crystal.
- Connect my Nano to it using ISP pins.
- Burn bootloader on it with "Bootloader: Yes"
- Put the ATmega8 in an UNO
- Open the blink sketch and click "Upload"
- I get the following error:

```
avrdude: Version 6.3-20201216
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

System wide configuration file is "C:\Users\Pouria\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.1/avrdude.conf"

Using Port                    : COM4
Using Programmer              : arduino
Overriding Baud Rate          : 38400
avrdude: stk500_getsync(): can't communicate with device: resp=0xfc

An error occurred while uploading the sketch
avrdude done.  Thank you.
```

I was using MiniCore v2.1.1 with Arduino IDE v1.8.16
Upgraded to v2.1.3 and still the same result.

Also, don't know if it's supposed to do this or not but after I burn bootloader the built-in LED keeps flashing. 3 fast flashes - 1sec off - 3 fast flashes - and so on

pouriap, 2022-02-08:
Oh and both with and without bootloader the sketch size is `6656` bytes. 

Perhaps the bootloader never gets uploaded? That would explain why the size doesn't change and the upload doesn't work either.

MCUdude, 2022-02-08:
> Also, don't know if it's supposed to do this or not but after I burn bootloader the built-in LED keeps flashing. 3 fast flashes - 1sec off - 3 fast flashes - and so on

This indicates that the bootloader has been loaded.

It's difficult for me to troubleshoot the issue when I can't reproduce the issues you're facing. All my Arduino cores should behave identically to MiniCore when uploading code, and this doesn't seem to be a common issue at all. My guess is that there's something wrong with your setup. Try uninstalling the boards manager version, clear the installation path, and try again.

BTW when did this become an issue for you? The feature where the bootloader is maintained when uploading using a programmer was added in v2.0.6 (one and a half years ago), and the "Bootloader" tools menu option in v2.0.0 (three years ago)

pouriap, 2022-02-08:
Oh OK. Will do a clean install then.

Just to make sure, it's supposed to show me a different code size when I change the bootloader from the menu right? I don't need to actually upload or burn the bootloader for the IDE to "understand" right?

> BTW when did this become an issue for you? The feature where the bootloader is maintained when uploading using a programmer was added in v2.0.6 (one and a half years ago), and the "Bootloader" tools menu option in v2.0.0 (three years ago)

IIRC the last time I used the bootloader programming was maybe 3-4 years ago. After that I rarely used the core or just used ISP programming. Recently I've been working with ATmega8, 6 months or so, and it's been like this ever since.

MCUdude, 2022-02-08:
Which IDE version are you using? I'm currently at 1.8.15.

> Just to make sure, it's supposed to show me a different code size when I change the bootloader from the menu right? I don't need to actually upload or burn the bootloader for the IDE to "understand" right?

Yes, it should use the correct flash size right away. It's pulled straight from boards.txt based on the menu option you selected.

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/boards.txt#L837-L850

pouriap, 2022-02-08:
Ok so I deleted everything and installed again.

Regarding the code size, I thought it's supposed to add the bootloader size to my sketch size so I was expecting for my sketch size to change. But after reading your comments more carefully I realized it's actually the maximum size that is supposed to change and it does indeed change. Not sure if it was like that before the delete.

Regarding programming, it still does not work in my UNO. But I tried programming it using an FTDI and it works. Again not sure if this is because of the delete because I don't think I had tried without the UNO board before.

So for some reason my UNO does not like anything other than a 328 inside it.

MCUdude, 2022-02-08:
> Regarding the code size, I thought it's supposed to add the bootloader size to my sketch size so I was expecting for my sketch size to change. But after reading your comments more carefully I realized it's actually the maximum size that is supposed to change and it does indeed change. Not sure if it was like that before the delete, but it probably was.

Correct. The bootloader has its own reserved space (of 512 bytes) at the very end of the flash memory. The compiled code size will stay the same regardless, but the "occupied" top 512 bytes has to be taken into account if using a bootloader.

> Regarding programming, it still does not work in my UNO. But I tried programming it using an FTDI and it works. Again not sure if this is because of the delete because I don't think I had tried without the UNO board before.

This is valuable information. It probably has something to do with the Arduino IDE recognizing the USB VID/PID on the UNO, and forces "default behavior" regardless. I have no idea why, and the UNO I used when I tested this is an old UNO R2, and I wasn't able to reproduce your issue. But if it works with a different USB to serial adapter I can safely tell that this is most likely not something I can easily fix, since it has to do with the IDE, not MiniCore.

pouriap, 2022-02-09:
Yes that was my impression too. I'll see if I can figure out how/why the IDE is recognizing the UNO board. My suspicion is the COM port because after installing the IDE its name in my system is changed to "Arduino UNO (COM4)".

Thanks again for your help and quick reply.

commit 6622e12e (2021-06-06): Update README.md

Closes #193