MCUdude/MiniCore #194: Add Optimize flag

pjbulls, 2021-06-06:
Would it be an idea to expose the compiler's optimization flags in a menu, similar to what is done in the Teensy core? Arduino always defaults to `-Os` which is decent, but I very rarely find myself running into space constraints especially on the 168/328. I put something like the following in boards.txt:

```
menu.Optimize=Optimization

# Compiler O flag
328.menu.Optimize.size=Default (size)
328.menu.Optimize.size.compiler.c.optimize_flag=-Os
328.menu.Optimize.size.compiler.cpp.optimize_flag=-Os
328.menu.Optimize.size.compiler.c.elf.optimize_flag=-Os

328.menu.Optimize.three=O3
328.menu.Optimize.three.compiler.c.optimize_flag=-O3
328.menu.Optimize.three.compiler.cpp.optimize_flag=-O3
328.menu.Optimize.three.compiler.c.elf.optimize_flag=-O3
...
```

The flag then also needs to be defined in platform.txt, obviously.

MCUdude, 2021-06-06:
The idea with MiniCore, and all my Arduino cores is to not expose _every_ setting as some other cores do. Most users are OK with -Os, so that's what I'll stick with. I also prefer to not add more tools menu options than we already have.
If you need other optimization flags you're free to use PlatformIO, which gives you full control over everything.

pjbulls, 2021-06-06:
Alright, fair enough. Agree that it's quite niche and possibly confusing, so I'll keep to my local modification 👍

closed 2021-06-06 by pjbulls