MCUdude/MiniCore #196: can't install versions >2.09 on linux

betyar, 2021-06-09:
I have several machines with both linux and windows, with MiniCore installed on all of them. I am unable to update or install MiniCore versions above 2.09 on any of my linux machines. This is the error I get:

>Tool avrdude is not available for your operating system.
>java.lang.RuntimeException: java.lang.Exception: Tool avrdude is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.jareva:179)
>	at java.lang.Thread.run(Thread.java:748)
>Caused by: java.lang.Exception: Tool avrdude is not available for your operating system.
>	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:88)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)

I have different versions of linux on some machines so I am sure it has nothing to do with the version of linux I am using. My windows machines don't have this problem, I was able to update them to the latest version of MiniCore. All of my machines use Arduino IDE version 1.8.15.

Is there any way around this?

MCUdude, 2021-08-15:
It looks like an IDE bug to be, but you can see the diff between 2.0.9 and 2.1.0 here:

https://github.com/MCUdude/MiniCore/compare/v2.0.9...v2.1.0

Basically there is a new millis and micros implementation, and two new programmers have been added.

MCUdude, 2021-12-19:
I'm closing this because it's not a MiniCore related bug, but rather an issue with the IDE itself, that has probably been fixed by now.

closed 2021-12-19 by MCUdude