MCUdude/MiniCore #197: ATmega328p bootloader burn error

zaffresky, 2021-06-15:
I am using an arduino uno board and have placed the new ATmega328chip in the socket as a simple setup. However, I am unable to burn the bootloader. I am using your minicore and selected the ATmega328 chip. I have tried using 8MHz internal crystal as well as 16 MHz external crystal but get the same error. I even tried using a different ATmega328p but the result was same. Here is the output:

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Administrator\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.2/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x03

avrdude done.  Thank you.

Error while burning bootloader.

```

Before trying your MiniCore I have also tried different tutorials which ask for external crystal and extra capacitors. It's been two days I am trying to make it work but sadly haven't managed it yet. Initially I suspected a faulty ATmega but even three other chips result in the same error. Could you please help me fix it? Thanks

MCUdude, 2021-06-15:
Hi!

The first thing you should do is to upen the IDE settings and turn on verbose upload. This makes the Avrdude output much richer and thus easier to figure out what's actually going on.

Are you able to burn the bootloader in the first place? You need a dedicated programmer such as the USBasp or Arduino as ISP. The output you posted looks to me like a "regular" upload where the USB to serial adapter can't communicate with the microcontroller.

So please connect an ISP programmer to the ATmega328PB, select the correct programmer in the Tools menu, and hit "Burn Bootloader", and then post the _entire_ output here.

zaffresky, 2021-06-15:
Hi,

Indeed with the verbose output it was much clear. 

It was perhaps my messy wiring on the breadboard. I wired it anew and it worked like a charm using Arduino as ISP.

Thanks for the pointer though :)

closed 2021-06-15 by zaffresky