MCUdude/MiniCore #198: Weird things on ATmega328PB

EG-Julien, 2021-06-24:
Hi,
I'm working on a new project based on an ATmega328PB-MU. I took the minimal setup schematic as reference but added an FT230XQ directly on my PCB.

Here is the schematic for my AT328PB _(Note that I replaced R5 with a 10k resistor)_ :
[image]And this is the schematic of my FTDI programmer :
[image]

I'm using an Arduino UNO to burn the bootloader through ICSP (MiniCore 2.1.2 & Arduino 1.8.15, 16 MHz, LTO dis, EEPROM retained, BOD 2.7V, Variant 238PB, bootloader listening on UART0) and it works fine. 
If I try to upload my sketch right after I burned the bootloader it uploads just fine on the MCU but I got some weird execution (like if the code was jumping some intructions). 
But, if I

 wait a while (~20s after burning bootloader) I can't upload any sketch on it ... got `avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync()` error (that normally comes when the MCU isn't responding to the programmer or false reset). I double checked my RST pin with my oscilloscope and it goes LOW as wanted when trying to program a MCU (got same scope result when it successfully uploads code and when it doesn't). I also checked FT230XQ by loop test and it works ust fine. So, I geuss, the problem isn't here (moreover it can't explain "random lines execution", if I may say so).

But I have no idea of what is going wrong ... Does anyone have an idea ?

If you need more details like designs files or whatever let me know.

Thanks for your help !

MCUdude, 2021-06-24:
Hi!

After you've burned the bootloader, does the LED attached to pin PB5 flash twice every second? If yes, that means that the bootloader is present and running. Does it also flash when you press and release the reset button?

Can you also make sure that the RTS pin on your USB to serial chip toggles when you try to upload?

If you upload code using your Uno, are you able to read and write to the serial monitor?

EG-Julien, 2021-06-25:
Hi!
Just tried with an another programmer (an another UNO board, I bought an Atmel programmer yesterday to try, but I didn't receive it yet). Got same results, after I've burned the bootloader PB5's led flash twice every second. It also flash when I reset the board. But I guess I find the problem (not the solution yet ...)

I did some measurement with my oscilloscope and the USB to serial chip toggles RST pin correctly (this is what I was guessing since the scope are correct for 328 version). And if I upload the code with the serial chip but I toggle the reset by hardware it works just fine. Same when I programmed a serial loop test on the MCU.

So now I know that's my RST system that doesn't work ... I don't even know why since I'm using a 10k resistor with a 100nF cap ... and used this chip with this system a hundred times but with ATmega328P. Is an incompatibility possible, isn't it ?

I got this, it fells below 2.7V but it seems not enough to toggle reset pin.
[image]

I firstly thought to replace the 10k R5 resistor with 100k one. It doesn't solve the problem even by pulling down the reset to 1.8V I got this:

[image]

And I found out why ... There is an incompatibility ... The ATmega328, 328P and 328PB version doesn't has the same reset threshold ([according to this guy](https://github.com/watterott/ATmega328PB-Testing/blob/master/README.md#reset) and the datasheet)... to use an FTDI 230XQ chip we should (didn't test yet) use a transistor to tied RST pin below 1.6V.

I'll try to find a solution to fix my FT230XQ and post it here. Since the issues isn't concerning you I'll close it up. 
Thanks for your help !

closed 2021-06-25 by EG-Julien

MCUdude, 2021-06-25:
Excellent research! I wasn't aware of the reset threshold level difference. I have designed a 328PB based board, but I've never had any issues with the auto-reset circuitry. But I've been using a CH340 based USB to serial chip, not an FTDI based one.

Can you try to replace the 100nF capacitor with a higher value one, for instance, 1uF or 10uF? That should pull the line down "harder", so it might actually work.

EG-Julien, 2021-06-25:
> Can you try to replace the 100nF capacitor with a higher value one, for instance, 1uF or 10uF? That should pull the line down "harder", so it might actually work.

I just tried with 1u, 10u and 47u (can't go further SMD 0603 ...) and it reach 1.8V but it's not enough yet. I'll have to find an another way.

MCUdude, 2021-06-25:
> Here is the result with a 10u. The reset time is reaching 90ms ...

Is this with a 10k or 100k pullup?

> I just tried with 1u, 10u and 47u (can't go further SMD 0603 ...) and it reach 1.8V but it's not enough yet. I'll have to find an another way.

Huh, that's strange! One option that will most likely work, but requires a PCB revision is to add a non-inverting buffer right after the capacitor and the pullup resistor. This way the buffer itself drives the reset pin, and as long as you connect your reset button to the input on the buffer, everything will work as normal. I use the 74LVC1G125 on the boards I sell because I was lucky to get a free, unused reel of them.

EG-Julien, 2021-06-25:
I tried with both and it doesn't change that much I'm reaching 1.75V ... not enough yet.

Yep this is what I'll do for the rev2!

I'll try to do a bridge divider to lower RTS MCU voltage to something like 3.5V or 4V I guess it could worked.

MCUdude, 2021-06-25:
Just out of curiosity, what does the RTS signal itself look like out of the FTDI chip? Does it go all the way down to ground when you try to upload?

EG-Julien, 2021-06-25:
Yes, it goes down during all of the time during programming period.

EG-Julien, 2021-06-25:
> I'll try to do a bridge divider to lower RTS MCU voltage to something like 3.5V or 4V I guess it could work.
So, it works but the MCU is randomly reseting itself and is very instable ... At least I tried ...