MCUdude/MiniCore #201: Sketch will not compile for Atmega8

kelogs, 2021-08-01:
It compiles fine for all the other boards in MiniCore.

The errors (partial output):

>    TCCR0A = 0;
>    ^~~~~~
>    TCCR1A
> eurosplit:308:3: error: 'TIMSK1' was not declared in this scope
>    TIMSK1 |= (1 << OCIE1A);                   //Set interrupt on compare match one channel; LED blink
>    ^~~~~~
> /home/kellogs/Sensory/trials/eurosplit/eurosplit.ino:308:3: note: suggested alternative: 'TIMSK'
>    TIMSK1 |= (1 << OCIE1A);                   //Set interrupt on compare match one channel; LED blink
>    ^~~~~~
>    TIMSK
> /home/kellogs/Sensory/trials/eurosplit/eurosplit.ino: In function 'void setup()':
> eurosplit:325:23: error: 'clock_div_4' was not declared in this scope
>    clock_prescale_set (clock_div_4);       // for 2 MHz operation, driving from the 8MHz RC oscillator (16 MHZ external xtal for Arduino)           TODO switch to div_4 in prod
>                        ^~~~~~~~~~~
> eurosplit:325:3: error: 'clock_prescale_set' was not declared in this scope
>    clock_prescale_set (clock_div_4);       // for 2 MHz operation, driving from the 8MHz RC oscillator (16 MHZ external xtal for Arduino)           TODO switch to div_4 in prod
>    ^~~~~~~~~~~~~~~~~~
> eurosplit:339:3: error: 'power_adc_disable' was not declared in this scope
>    power_adc_disable();
>    ^~~~~~~~~~~~~~~~~
> /home/kellogs/Sensory/trials/eurosplit/eurosplit.ino:339:3: note: suggested alternative: 'power_all_disable'
>    power_adc_disable();
>    ^~~~~~~~~~~~~~~~~
>    power_all_disable
> eurosplit:340:3: error: 'power_spi_disable' was not declared in this scope
>    power_spi_disable();
>    ^~~~~~~~~~~~~~~~~
> /home/kellogs/Sensory/trials/eurosplit/eurosplit.ino:340:3: note: suggested alternative: 'power_all_disable'
>    power_spi_disable();
>    ^~~~~~~~~~~~~~~~~
>    power_all_disable
> exit status 1
> 'digitalPinToPCMSK' was not declared in this scope

The includes from .ino sketch:

```
#include <avr/io.h>
#include <avr/sleep.h>
#include <avr/power.h>
#include <avr/wdt.h>
//#include <Arduino.h>
#include <EEPROM.h>
```

[image]

- Arduino IDE 1.8.12
- Minicore 2.1.3
- Linux kellogs-laptop 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) x86_64 GNU/Linux

I don't suppose this happens a lot... any tips ?

Thanks

MCUdude, 2021-08-01:
The ATmega8 is older than all the other targets and has a different timer0 configuration. The code you're trying to compile for the ATmega8 won't work because it has different timer registers than the newer chips.

You'll have to rewrite your code to with for the ATmega8. If you're not able to do this, you can use an ATmega88 for instance, which is 100% compatible with the ATmega328P in terms of registers.

kelogs, 2021-08-01:
Wow, that was fast!

Ok, perhaps I can rewrite that part.

How about 

>   eurosplit:340:3: error: 'power_spi_disable' was not declared in this scope

Is that somehow also related  to the timer ?

MCUdude, 2021-08-01:
> Wow, that was fast!

I got an email notification, and I was already on my computer 😉 

>> eurosplit:340:3: error: 'power_spi_disable' was not declared in this scope

> Is that somehow also related to the timer?

This is because the ATmega8 doesn't have a power save register for the SPI peripheral.

From the power.h library:

```c
#if defined(__AVR_HAVE_PRR_PRSPI)
#define power_spi_enable()      (PRR &= (uint8_t)~(1 << PRSPI))
#define power_spi_disable()     (PRR |= (uint8_t)(1 << PRSPI))
#endif
```

`__AVR_HAVE_PRR_PRSPI` does not exist in the ATmega8 definitions file (iom8.h), and as far as I know, the ATmega8 doesn't have a single power reduction register! If power usage is a concern, apparently the ATmega8 is not the right choise.

kelogs, 2021-08-01:
I see, OK then, I am off to solder another MCU in place of Atmega8.

Thank you very much!

MCUdude, 2021-08-01:
> I see, OK then, I am off to solder another MCU in place of Atmega8.
Thank you very much!

You're welcome!

closed 2021-08-01 by MCUdude