MCUdude/MiniCore #203: atmega 8  rcswitch

tamer1526, 2021-09-08:
atmega8 not working with rcswitch library tested

MCUdude, 2021-09-08:
That's not much of a clue for me to work on. You need to give me the URL where I can fin the library, ideally post your code and the error message the compiler gives you.

tamer1526, 2021-09-08:
the link of the library https://github.com/sui77/rc-switch
the code demo simple reciver not working for atmega8 and it work perfect for atmega328
  if (rfc.available()) {
    Serial.print("Received ");
    Serial.print( rfc.getReceivedValue() );
    Serial.print(" / ");
    Serial.print( rfc.getReceivedBitlength() );
    Serial.print("bit ");
    Serial.print("Protocol: ");
    Serial.println( rfc.getReceivedProtocol() );
    rfc.resetAvailable();
  }

MCUdude, 2021-09-08:
Are you getting an error from the compiler, or does it compile just fine, but doesn't work?

tamer1526, 2021-09-08:
no compiling errors but it doesnt work

MCUdude, 2021-09-08:
I don't have any rc-switch compatible hardware, so I'm not able to test it with real hardware. However, not all libraries will work with aTmega8 because the author hasn't designed the library to work with it, You should open an issue over at the rc-switch Github repo and explain what issues you're facing and how they can reproduce the issue. They might be able to help you.

closed 2021-12-19 by MCUdude