MCUdude/MiniCore #204: Error compiling U8GLIB for board ATmega328PB

maikodesousa, 2021-09-11:
I made a board with 328PB and burned the MiniCore on it. The IDE does not support compiling code for U8Glib with MiniCore when I use 328PB, but it works fine if I switch to 328P.

The problem apparently is with the SPI, but the display is I2C. Anyone else with this problem?

**This is the code:**

#include "U8glib.h"
float numero = 0;
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NONE|U8G_I2C_OPT_DEV_0);	// I2C / TWI 
void draw(void) {
  u8g.setFont(u8g_font_9x15B);
  u8g.drawStr( 0, 20, "Numero");
  u8g.setPrintPos(0, 50);
  u8g.print(numero);
}
void setup () {
Serial.begin (9600);
}

void loop () {
  if (Serial.available ()) 
    {
      numero = Serial.read ();
    }
  u8g.firstPage();  
    do {
      draw();
    } 
  while( u8g.nextPage() );
  delay(50);
}

**And this is the message from IDE:**

C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c: In function 'u8g_atmega_spi_out':
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c:76:3: error: 'SPDR' undeclared (first use in this function); did you mean 'SPDR0'?
   SPDR = data;
   ^~~~
   SPDR0
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c:76:3: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c:78:12: error: 'SPSR' undeclared (first use in this function); did you mean 'SPDR'?
   while (!(SPSR & (1<<SPIF)))
            ^~~~
            SPDR
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c: In function 'u8g_com_atmega_hw_spi_fn':
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_hw_spi.c:118:7: error: 'SPCR' undeclared (first use in this function); did you mean 'SPCR0'?
       SPCR = 0;
       ^~~~
       SPCR0
"C:\\Users\\Maiko\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR "-IC:\\Users\\Maiko\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\Maiko\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\pb-variant" "-IC:\\Users\\Maiko\\Documents\\Arduino\\libraries\\U8glib\\src" "C:\\Users\\Maiko\\Documents\\Arduino\\libraries\\U8glib\\src\\clib\\u8g_com_atmega_st7920_spi.c" -o "C:\\Users\\Maiko\\AppData\\Local\\Temp\\arduino_build_904697\\libraries\\U8glib\\clib\\u8g_com_atmega_st7920_spi.c.o"
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c: In function 'u8g_atmega_st7920_hw_spi_shift_out':
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c:75:3: error: 'SPDR' undeclared (first use in this function); did you mean 'SPDR0'?
   SPDR = val;
   ^~~~
   SPDR0
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c:75:3: note: each undeclared identifier is reported only once for each function it appears in
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c:77:12: error: 'SPSR' undeclared (first use in this function); did you mean 'SPDR'?
   while (!(SPSR & (1<<SPIF)))
            ^~~~
            SPDR
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c: In function 'u8g_com_atmega_st7920_hw_spi_fn':
C:\Users\Maiko\Documents\Arduino\libraries\U8glib\src\clib\u8g_com_atmega_st7920_hw_spi.c:135:7: error: 'SPCR' undeclared (first use in this function); did you mean 'SPCR0'?
       SPCR = 0;
       ^~~~
       SPCR0
Using library U8glib at version 1.19.1 in folder: C:\Users\Maiko\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board ATmega328.

MCUdude, 2021-09-11:
The u8glib is outdated and does not support ATmega328PB, which has different names for its SPI registers than the 328P has. Use the u8g2 library instead.

maikodesousa, 2021-09-11:
> The u8glib is outdated and does not support ATmega328PB, which has different names for its SPI registers than the 328P has. Use the u8g2 library instead.

Using u8g2 library instead u8glib, I have a similar problem. It compiles for 328P, but not for 328PB.

C:\Users\Maiko\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp: In function 'uint8_t u8x8_byte_arduino_2nd_hw_i2c(u8x8_t*, uint8_t, uint8_t, void*)':
C:\Users\Maiko\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:1354:7: error: 'Wire1' was not declared in this scope
       Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
       ^~~~~
C:\Users\Maiko\Documents\Arduino\libraries\U8g2\src\U8x8lib.cpp:1354:7: note: suggested alternative: 'Wire'
       Wire1.write((uint8_t *)arg_ptr, (int)arg_int);
       ^~~~~
       Wire
Using library SDL_Arduino_INA3221-master in folder: C:\Users\Maiko\Documents\Arduino\libraries\SDL_Arduino_INA3221-master (legacy)
Using library Wire at version 1.0 in folder: C:\Users\Maiko\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3\libraries\Wire 
Using library U8g2 at version 2.30.1 in folder: C:\Users\Maiko\Documents\Arduino\libraries\U8g2 
Using library SPI at version 1.0 in folder: C:\Users\Maiko\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3\libraries\SPI 
exit status 1
Error compiling for board ATmega328.

fcig, 2021-11-07:
I have the same error here using u8g2 and 328PB: "'Wire1' was not declared". The strange thing is that in the past I could upload my sketch successfully to 328PB so I tried to remember what I did that could cause this issue and I remembered that I updated MiniCore to version 2.1.3. With that in mind I installed MiniCore 2.1.2 and it I could compile my skech successfully again to 328PB!

MCUdude, 2021-11-28:
Sorry for the late reply. Can someone provide an example sketch I can try so I can reproduce the issue?

MCUdude, 2022-03-07:
Uncomment the following line in U8x8lib.h, and it will work.

https://github.com/olikraus/u8g2/blob/b50b8fcabfd3c6534b600f5fd169e104571d3408/cppsrc/U8x8lib.h#L124

closed 2022-03-07 by MCUdude