MCUdude/MiniCore #208: What is the right programmer in the Arduino IDE?

RudolfAtHome, 2021-11-13:
Your "Getting started..." has the line 
**Select what kind of programmer you're using under the Programmers menu.**
What is the right programmer from the huge list? I've tried a few, all of them end in timeout.
Alernatively I used the standard ATMega Nano: the upload works fine without any problem. But there is no support for ATMega328PB.

MCUdude, 2021-11-13:
Use the one you like and that works for you. I prefer to use the USBasp or USBtinyISP.

> I've tried a few, all of them end in timeout.

I might be able to help you out if you post the actual error message you're getting. Remember to turn on verbose upload in the IDE settings.

> Alernatively I used the standard ATMega Nano: the upload works fine without any problem. But there is no support for ATMega328PB.

The Arduino as ISP should support any AVR that's programmed using an ISP interface. What's the error you're getting?

RudolfAtHome, 2021-11-13:
Here is the error Message from Arduino IDE 1.8.16:

"C:\\Users\\Rudol\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" 
	-A "C:\\Users\\Rudol\\AppData\\Local\\Temp\\arduino_build_881376/ReadSignature.ino.elf"
Sketch uses 5472 bytes (16%) of program storage space. Maximum is 32256 bytes.
Global variables use 533 bytes (26%) of dynamic memory, leaving 1515 bytes for local variables. Maximum is 2048 bytes.
C:\Users\Rudol\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino18/bin/avrdude 
	-CC:\Users\Rudol\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/avrdude.conf -v -V -patmega328pb -carduino 
	-PCOM3 -b115200 -D -Uflash:w:C:\Users\Rudol\AppData\Local\Temp\arduino_build_881376/ReadSignature.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Rudol\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x32
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x32
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x32
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xfc
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xfc
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xfc
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x1c
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xe0
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x1c

avrdude done.  Thank you.

Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.

MCUdude, 2021-11-13:
That error message indicates that you're not trying to burn the bootloader, but you're trying to update using a USB to serial connection of some sort. The ATmega328PB needs a bootloader first.

RudolfAtHome, 2021-11-13:
Hallo Hans,

 

Ihr Name lässt hoffen, dass Sie typisch Deutsch sprichst!? Vielen Dank für die beiden schnellen Antworten.

 

Ich muss mich wohl durchringen, den Bootloader zu brennen. Gibt es dazu eine Anleitung, oder geht das „wie immer“ über den ISCP-Block?

 

Mit freundlichem Gruß

Rudolf Schenke

MCUdude, 2021-11-13:
> Ihr Name lässt hoffen, dass Sie typisch Deutsch sprichst!? Vielen Dank für die beiden schnellen Antworten.

Sorry, I'm not very good at German at all. We can do Norwegian if you prefer this over English 😉 

> Ich muss mich wohl durchringen, den Bootloader zu brennen. Gibt es dazu eine Anleitung, oder geht das „wie immer“ über den ISCP-Block?

I'm not sure what your hardware looks like, but if it is a complete 328PB based board with a 2x3 pin ISP header on, you just select your preferred settings in the IDE (Clock speed, brown-out detection, EEPROM save, etc.). Then you'll have to choose the correct programmer under the "Programmers" sub-menu and then run "Burn Bootloader". Same procedure as on any other AVR-based board.

harryberlin, 2021-11-14:
Den Bootloader kannst nur mit nen Programmer über 6pin ISP-Port brennen.
Und auch nur mit dem Button "Bootloader brennen". Mit Sketch Upload geht es nicht.

closed 2022-08-19 by MCUdude