MCUdude/MiniCore #209: Question: is there a way to get MCU clock type and BOD status from precompiller ?

Demonic098, 2021-11-19:
My project is trgeted for specific MCU configuration and can act werdly if configured incorrectly. 
There is a simple way to get MCU frequency for any core/MCU but i need to check MCU clock type(Internal/external), BOD status  and if core is set to Minicore from precompiller.
Currently i am using following code:
#if (F_CPU == 8000000UL)
#else 
#error "Set (CPU clock to 8MHz Internal) (BOD to disable ) and BURN MiniCore bootloader" 
#error "https://github.com/MCUdude/MiniCore"
#endif

Demonic098, 2021-11-19:
Update: i've found MINICORE definition.
```
#if defined (MINICORE) && (F_CPU == 16000000UL)
```

MCUdude, 2021-11-28:
There are no macros that can indicate that the fuse bits aren't correct. Bear in mind that the compiler doesn't really know anything about the fuse bits. It's a separate thing that Avrdude takes care of.

I could technically add "defines" for the various settings, but it wouldn't necessarily work well because there's no way to tell the compiler which settings are actually on the microcontroller.

Demonic098, 2021-11-29:
It's not about microcontroller fuse bits. I only care if MINICORE MCU settings are set correctly.

MCUdude, 2022-03-07:
> I only care if MINICORE MCU settings are set correctly.

This is not currently possible.

closed 2022-08-19 by MCUdude