MCUdude/MiniCore #211: atmega328B runs too slow

wa2mze, 2021-11-30:
I wired up the atmega328B on a breadboard with a 16mhz crystal.  I tried the blink sketch (used D23 as LED pin), and programmed it with no bootloader using an atmel-ice.  I set the clock at 16mhz external.  It compiled and loaded, but it's blinking about 10-15 times TOO SLOW with delay(1000) selected.  I tried internal clock rates of 1-8mhz, and here it seems to be blinking almost at the expected 1/2 hz rate.  Thinking I had a bad crystal, I removed the 16mhz one and soldered in a 20mhz crystal, and selected 20mhz external clock.  Once again, at least a magnitude or more order too slow!
I've never seen this happen before.  LTO was enabled.

copied avrdude output.  I can't tell if the fuses are being set correctly!
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/home/ken/.arduino15/packages/MiniCore/hardware/avr/2.1.3/avrdude.conf"
         User configuration file is "/home/ken/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : atmelice_isp
avrdude: usbdev_open(): Found Atmel-ICE CMSIS-DAP, serno: J41800120684
avrdude: max packet size expected 912, but found 512 due to EP 0x01's wMaxPacketSize
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : JTAG3_ISP
         Description     : Atmel-ICE (ARM/AVR) in ISP mode
         Vtarget         : 5.0 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/tmp/arduino_build_788772/Blink.ino.with_bootloader.hex"
avrdude: writing flash (1204 bytes):

Writing | ################################################## | 100% 0.33s

avrdude: 1204 bytes of flash written
avrdude: verifying flash memory against /tmp/arduino_build_788772/Blink.ino.with_bootloader.hex:
avrdude: load data flash data from input file /tmp/arduino_build_788772/Blink.ino.with_bootloader.hex:
avrdude: input file /tmp/arduino_build_788772/Blink.ino.with_bootloader.hex contains 1204 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.35s

avrdude: verifying ...
avrdude: 1204 bytes of flash verified

avrdude done.  Thank you.

MCUdude, 2021-11-30:
You need to set the correct settings in the Tools many and click "Burn Bootloader" to set the fuses, even though you're not using a bootloader. Incorrect speed is the first sign of incorrect fuse settings.

MCUdude, 2021-12-19:
So, did it work ut for you?

closed 2022-03-07 by MCUdude