MCUdude/MiniCore #213: Announcement: debugWIRE hardware debugger for (almost) all MiniCore MCUs ready

felias-fogg, 2022-01-17:
Hi,

I just wanted to announce that I finished an Arduino sketch called [dw-link](https://github.com/felias-fogg/dw-link), which you can use to turn you Uno (or equivanlent) into a hardware debugger. It will empower you to use avr-gdb on a host to debug any Arduino sketch on one of the MiniCore MCUs. That is except for ATmega8, which does not support debugWIRE, and except for ATmega48 and 88 without any suffix, which  have stuck-at-one bits in their program counter---the A's and PA's work, however.

I wanted to put the announcement into the Arduino forum, but it is closed. Perhaps, one should open a _discussion_ for the MiniCore Github rep here.

MCUdude, 2022-01-18:
Wow, this is very impressive, I'll definitely give this a try!

Does this mean we finally have support for "real" hardware debugging in PlatformIO? Hardware debugging is I'm my opinion the "only thing left" before PlatformIO does everything I need it to do.

Out of curiosity, could this be accomplished with an official AVR debugger that already "speaks" debugWire? The reason I'm asking is that it would be really handy to have support for official AVR debuggers like the jtagmkii and the Atmel ICE, since it would allow debugging of _any_ "classic" AVR (that has either dW or JTAG).

> except for ATmega48 and 88 without any suffix, which have stuck-at-one bits in their program counter---the A's and PA's work, however.

Is there a silicon bug in these parts that prevents debugging from working on these parts?

felias-fogg, 2022-01-18:
> Does this mean we finally have support for "real" hardware debugging in PlatformIO? Hardware debugging is I'm my opinion the "only thing left" before PlatformIO does everything I need it to do.

Yes, you also find the necessary platformio.ini file in the example folder.

> Out of curiosity, could this be accomplished with an official AVR debugger that already "speaks" debugWire? The reason I'm asking is that it would be really handy to have support for official AVR [...]

Unfortunately not. I guess, the way to use the official debuggers is to use AVaRICE, which I was not able to get running.

> 
> > except for ATmega48 and 88 without any suffix, which have stuck-at-one bits in their program counter---the A's and PA's work, however.
> 
> Is there a silicon bug in these parts that prevents debugging from working on these parts?

I think, it is an "undocumented silicon feature". Atmel had documented it for the ATmega16: "If software reads the Program Counter from the Stack after a call or an interrupt, unused bits (15:13) should be masked out" (p. 12). It is not a problem for the MCU itself and also Atmel-ICE/MPLAB is able to deal with it (by masking or setting the bits). GDB gets confused when it finds PC values on the stack that point to unknown locations, though.

MCUdude, 2022-08-19:
I'm going through old issues, and I'm closing this one as it's an announcement and not really a MiniCore-related issue.
But thanks for your contribution!

closed 2022-08-19 by MCUdude