MCUdude/MiniCore #214: attachInterrupt and detachInterrupt compiler errors in added .cpp file

Ziggy2013, 2022-01-29:
An Atmega328 sketch using an added .cpp file with attachInterrupt(....) and detachInterrupt(...)  statements compiles successfully in the Arduino IDE 1.8.6 with no errors.

However the same sketch gives compiler errors on all attachInterrupt and detachInterrupt statements when minicore is set in the Tools menu.

An example of the minicore error is

**sketch\PCB_lin_read_01.cpp: In function 'void pin_delimitbit_int()':

PCB_lin_read_01.cpp:192:41: error: invalid conversion from 'void (*)()' to 'uint8_t {aka unsigned char}' [-fpermissive]

       detachInterrupt(pin_delimitbit_int);**

If the same code is in the .ino file, attachInterrupt and detachInterrupt statements appear to compile without minicore errors. 

I suspect the issue is related to the use of #include <arduino.h> in the added .cpp file but have not been able to confirm that.

MCUdude, 2022-01-29:
The official Arduino core has the -fpermissive flag set, but MiniCore doesn't. My advice is to fix the initial problem. And it's always a good idea to turn on compiler warnings in the IDE settings.

> The -fpermissive flag causes the compiler to report some things that are actually errors (but are permitted by some compilers) as warnings, to permit code to compile even if it doesn't conform to the language rules. You really should fix the underlying problem. Post the smallest, compilable code sample that demonstrates the problem.

>> -fpermissive
Downgrade some diagnostics about nonconformant code from errors to warnings. Thus, using -fpermissive will allow some nonconforming code to compile.

Ziggy2013, 2022-01-29:
Hi and thanks for the quick response.

The code is not my code but I have stripped everything out to the demonstrate the issue

main.ino file is

void setup() {
}

void loop(){
}

The added file test.cpp is

#include <arduino.h>

void pin_delimitbit_int()
{
      detachInterrupt(pin_delimitbit_int);   
}

When compiled in minicore atmega328 the compiler error is

In function 'void pin_delimitbit_int()':
PCB_lin_read_01.cpp:7:41: error: invalid conversion from 'void (*)()' to 'uint8_t {aka unsigned char}' [-fpermissive]
       detachInterrupt(pin_delimitbit_int);

Compiles ok in the Arduino IDE 1.8.6

MCUdude, 2022-01-29:
Well, `detachedInterrupt()` only takes an uint8_t as a parameter, not a function pointer like you're trying to pass is not allowed and should cause an error. Pass the interrupt number you're trying to disable instead.
MiniCore is just preventing you from forcing a cube into a round hole.

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/cores/MCUdude_corefiles/WInterrupts.c#L241

If you're interested, here's a bit of history regarding Arduino and the -fpermissive flag. It was a big mistake adding it to the official Arduino core, and I refuse to do the same. https://github.com/arduino/ArduinoCore-avr/issues/268

Ziggy2013, 2022-01-29:
I read the history.   -fpermissive  should definitely be removed from the Arduino core. Serious errors like this detachInterrupt parameter should not just slip through as warnings.

This episode has been a lesson.  Unfortunately I am now going to have to thoroughly understand some borrowed code I had expected to just be able to reuse.

Thanks for your help.

MCUdude, 2022-01-29:
> I read the history. -fpermissive should definitely be removed from the Arduino core. Serious errors like this detachInterrupt parameter should not just slip through as warnings.

Yes, I totally agree. I usually like to know what's going on, so I've turned on all compiler warnings too. I believe the Arduino developers disabled all warnings by default to make the IDE more "pleasing" and less "noisy" for new users. But the compilers give you warnings for a reason!

As for the code that didn't compile with MiniCore, I highly doubt it even would work with the official Arduino core either. The detachInterrupt pin expects an interrupt number (0 or 1), so the function passed to detachInterrupt has to be located in memory address 0x0000 or 0x0001. That's very unlikely.

Thanks for your understanding and for using MiniCore! I'll close this issue now, but don't be afraid to ask more MiniCore related questions here, I'll still get notified.

closed 2022-01-29 by MCUdude