MCUdude/MiniCore #215: Atmega328pb bootloader

ITstreet1, 2022-01-31:
I have a project where I use Atmega328p MCUs. As these shortages step in, I buy what can be bought. Atmega328pb for instance...
Here comes the problem.

My bad as I didn't check the pinout and the datasheet. Just solder it on and tried to burn the bootloader.
The board I used is the board that several 328p runs just fine. There is a 16MHz crystal, with 22pF caps. But there are VCC and GND on pins 3 and 6. On 328pb there are I2C1 pins. When trying to burn the bootloader using an Uno as an ISP and miniCore Atmega328pb in settings I get this error:

Arduino: 1.8.15 (Windows 10), Board: "ATmega328, Yes (UART0), EEPROM retained, 328PB, BOD 2.7V, LTO disabled, External 16 MHz"

```

avrdude: Version 6.3-20201216

         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/

         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.1/avrdude.conf"

         Using Port                    : COM5

         Using Programmer              : stk500v1

         Overriding Baud Rate          : 19200

         AVR Part                      : ATmega328PB

         Chip Erase delay              : 9000 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : dedicated

         RETRY pulse                   : SCK

         serial program mode           : yes

         parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         ByteDelay                     : 0

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                  Block Poll               Page                       Polled

           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff

           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff

           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

[... 234 lines cut ...]
avrdude: input file C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.1/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex contains 32768 bytes

avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...

avrdude: verification error, first mismatch at byte 0x7e00

         0x00 != 0x01

avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

```

Should I cut those traces? Did I bricked the MCU? Can I use this MCU as a drop in replacement for a 328p?

MCUdude, 2022-01-31:
I would perhaps first try to replace those 22pF caps with 10 or 15 pF ones.

Long story short, The 328PB's bigger brother is the ATmega324PB, and these are very similar internally. A major drawback to these parts is that they don't have any crystal "full swing oscillator" option anymore, which means that their external clock circuity may be very picky and can randomly stop working or not matched correctly, sometimes even weeks of operation.

A while ago I designed a commercial board where a 324PB was chosen because it had more IO than 324P. I used the same external oscillator as I always do, and there were 100 boards manufactured. All of these had to be recalled to have their external clock circuitry replaced. Once I figured out the correct values, it has never been an issue anymore.

If new capacitors don't work, try using a different programmer type, for instance, a USBasp or USBtinyISP. The additional pins shouldn't cause an issue as long as they're not configured as outputs. I do have an Atmega328PB based board, and I've never had any issues bootloading it.

MCUdude, 2022-02-08:
Did you figure it out?

ITstreet1, 2022-02-23:
Ok, I tried 10pF and 15pF.
With Arduino as ISP I ended the same as above.

While using USBTiny I get this:

```
avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Setting SCK period to 32 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: Setting SCK period to 32 usec
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x3f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```
I will try USBAsp later

ITstreet1, 2022-02-23:
This is from USBAsp:

```

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 32.0
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: set SCK frequency to 16000 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: set SCK frequency to 16000 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbasp
         Setting bit clk period        : 32.0
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
[... 99 lines cut ...]
avrdude: set SCK frequency to 16000 Hz
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex"
avrdude: writing flash (32768 bytes):

Writing | ################################################## | 100% -0.00s

avrdude: 32768 bytes of flash written
avrdude: verifying flash memory against C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex:
avrdude: load data flash data from input file C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex:
avrdude: input file C:\Users\dekip\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3/bootloaders/optiboot_flash/bootloaders/atmega328pb/16000000L/optiboot_flash_atmega328pb_UART0_115200_16000000L_B5.hex contains 32768 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7e00
         0xff != 0x01
avrdude: verification error; content mismatch

avrdude done.  Thank you.

Error while burning bootloader.

```

Mylossh, 2022-02-25:
Hi everybody!
I'd like to hook up to the topic.

Here's my setup that works just normal for burning bootloader by using MiniCore Atmega328, but with internal crystal:
Internal 8MHz clock, BOD disabled, EEPROM retained, LTO disabled, 328PB, Bootloader UART0;

Goal is to obviously get it to work with external 16MHz crystal which doesn't in my case too. I've tried with normal 2pin crystal package and 22pF as well as 6.8pF caps. Doesn't work. Now I'm planning to try with different crystal type and load caps.

Any other suggestions?

JoaquimBoavida, 2022-03-03:
Hi Mylossh

I had similar problems with mega328pb using resonators. Mega328p worked fine for years, switch to m328pb (same PCB), and the problems started. Problems while programming with AVRISP MKII and stability while operating.
I had some contacts with Microchip tech support but with no luck. Dead end. 
Eventually the problem was solved by redesigning the PCB  and use a oscillator on XTAL1 pin.
No more issues.
HTH 

Joaquim

MCUdude, 2022-03-03:
> Eventually the problem was solved by redesigning the PCB and use a oscillator on XTAL1 pin.

Yes, using a dedicated clock connected to the XTAL1 pin would definitely solve the issue. Or create your own crystal driver like shown below (where IC2A is optional):

[image]

ITstreet1, 2022-03-03:
That looks interesting...

MCUdude, 2022-03-03:
I'm using this exact circuit to drive crystals when using ATtiny13 or ATmega4809, where neither of them can drive a crystal but supports a clock input signal.

ITstreet1, 2022-03-03:
Can you put some light on this 74hc04 IC? From what I can see it is Schmitt Trigger. According to the datasheet https://www.diodes.com/assets/Datasheets/74HC04.pdf, it seems to have 3 channels. In this schematic you provided, you use two of the ICs and only one channel from both. Besides, You use pins 2 and 4, and for input and output, and it should be, say 1A and 1Y (pins 1 and 2).
Correct me where I am wrong.

MCUdude, 2022-03-03:
Don't look too much on the actual pins. I've been using a 74HC1G04 in a SOT353 package.

The 74HC04 does not have a Schmitt triggered input, the 74HC14 does.

ITstreet1, 2022-03-03:
https://assets.nexperia.com/documents/data-sheet/74HC_HCT1G04.pdf
This one? That makes things clear.

ITstreet1, 2022-03-03:
What could be used as a replacement? This is not available at my place. :(
I can order it abroad, though.

JoaquimBoavida, 2022-03-03:
I use this (or similar):

https://eu.mouser.com/ProductDetail/Epson-Timing/SG5032CCN-16000000M-HJGA3?qs=sGAEpiMZZMvjAcTDbo5QTpwxKcangighf21RCgxfPBQ%3D

MCUdude, 2022-03-03:
You can use any 74XX04, It's a generic, non-Schmitt-triggered inverter. if there are unused inverters in a package, you can just wire the inputs low or high, and leave the output unconected.

Mylossh, 2022-03-13:
Hello guys! 
Since I have around 20pcs of 328 in PB version, I've decided to design a small dev board for it, so that I can try if it accepts bootloader with oscillator instead of crystal as you suggested. 
Could anyone possibly check schematic from PDF that is attached? I'm not sure if everything is connected as it should be. If I understood the Datasheet, oscillator output should be connected to PB6/XTAL1? 
Thanks in advance! 
[AtMega328PB dev.pdf](https://github.com/MCUdude/MiniCore/files/8240489/AtMega328PB.dev.pdf)

MCUdude, 2022-03-13:
The schematic looks alright to me. I would however recommend you to add a diode from the RST line to VCC, to prevent voltage doubling when the DTR signal does its thing, which can make the chip enter high voltage parallel programming mode.

Mylossh, 2022-03-13:
Thank you for a check and suggestion Hans! I will add it.

MCUdude, 2022-08-19:
@ITstreet1 did you resolve the issue in the end?

If not, as a test, you can try [Adafruits Arduino as ISP sketch](https://github.com/adafruit/Standalone-Arduino-AVR-ISP-programmer) instead. It works exactly like the "original" one, when used with a 16 MHz UNO, it outputs a stable, 8 MHz clock on digital pin 9. If you connect this pin directly to the XTAL1 pin on your 328PB, it _should_ work.

ITstreet1, 2022-08-19:
Mate, I just forget this one. I will give it a try.
Thank you.

chiunit, 2022-09-22:
Hello my dear friend, your mini core really saved me a lot and I was able to solve the boot loading, but I have an issue. I interfaced it with and lcd and sometimes it just keeps restarting and shows random letters and sometimes it just restarts repeatedly. I’m using Atmega328pb

mcuee, 2022-12-23:
In any case, MiniCore optiboot_flash bootloader works fine with ATmega328PB. So this issue can be closed.

closed 2022-12-23 by MCUdude

ITstreet1, 2022-12-26:
Gotta open this one. 

Finally, I find time to make this one work. Board is here, and everything is soldered. I successfully burned the bootloader with Nano. Tried to upload a simple Blink sketch over the USB (UART CH340C) but get this error:

> 
> Arduino: 1.8.19 (Windows 10), Board: "ATmega328, Yes (UART0), EEPROM retained, 328PB, BOD 2.7V, LTO disabled, External 16 MHz"
> 
> Sketch uses 1232 bytes (3%) of program storage space. Maximum is 32256 bytes.
> 
> Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x84
> 
> avrdude: stk500_recv(): programmer is not responding
> 
> avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x84
> 
> Problem uploading to board.  See https://support.arduino.cc/hc/en-us/sections/360003198300 for suggestions.
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.

What is interesting, I can upload a sketch using Nano as a programmer over an SPI. I double-checked all solder joints, and everything is fine. I also tried your "diode fix" from the above post, but no luck.

[Schematic.pdf](https://github.com/MCUdude/MiniCore/files/10304570/Schematic.pdf)

Any suggestion?