MCUdude/MiniCore #217: Bootloader incompatibility

marko-pi, 2022-02-20:
Perhaps this is more a question than an issue, but I did not know of any other way to contact you.

I noted that if I upload _standard_ Optiboot to my ATmega328P using **Arduino AVR Boards** / **Arduino Pro or Pro Mini**, I cannot upload sketches using your MiniCore board manager.  Vice-versa is also true, if I upload _customised_ Optiboot to my ATmega328P using **MiniCore** / **ATmega328P** (5V/external oscillator), I cannot upload sketches using standard Arduino board manager.  In particular, I get bunch of "not in sync" errors.

Could you be more specific about the difference between two Optiboots, because I have to decide using one of them, so I will avoid any further confusion.  Which one should I use in your opinion?

MCUdude, 2022-02-20:
**Arduino Pro or Pro Mini** doesn't use Optiboot at all, but an older bootloader called ATmegaBOOT. The Pro Mini is programmed at 57600 baud while MiniCore uses Optiboot and 115200 baud at 16 MHz. Optiboot is superior in any way. It's faster, smaller and doesn't rely on the internal watchdog timer, so this can use used in your sketch.

The Arduino UNO uses the Optiboot bootloader and 115200 baud, so this is compatible with MiniCore.

https://github.com/arduino/ArduinoCore-avr/blob/b3243815e2ed8c72f9ac3669e69632a590b1d048/boards.txt#L861

marko-pi, 2022-02-20:
Oh, that makes sense!  I wasn't aware that Arduino Pro or Pro Mini uses something older than Optiboot.

So if I understand you right, your optimized Optiboot is compatible with standard Optiboot regarding sketch UART upload?

In that case I will use MiniCore exclusively for my chips from now on.

MCUdude, 2022-02-20:
> So if I understand you right, your optimized Optiboot is compatible with standard Optiboot regarding sketch UART upload?

Absolutely! The bootloader is one thing, but the MiniCore source code is more optimized and efficient than the official Arduino AVR core. The only difference is that link time optimizer (LTO) is enabled my default by the Arduino AVR core, but disabled by default on MiniCore. Enable this, and pretty much any sketch will be smaller using MiniCore.

> Maybe I should even overwrite bootloader for my Arudino Pro Minis with MiniCore?

Yes, you'll have to do this if you want to upload using a USB to serial adapter on your Pro Minis using MiniCore.

marko-pi, 2022-02-20:
Thanks a lot!

closed 2022-02-20 by marko-pi