MCUdude/MiniCore #219: Fuses - 8 vs 4 MHz internal osc

djwhouse, 2022-03-06:
I am confused a bit on the MiniCore fuse settings. I don't see a difference in the fuse settings between the 8MHz and the 4MHz internal osc settings for a AtMega328p.   How does the miniCore get the variations of 8, 4, and 2 MHz?

I'm an EE Engineer, retired. 

If not in the fuses, what sets the divide of the 8MHz? I see the 1MHz fuse variation. 

I am missing something :-)

MCUdude, 2022-03-06:
The trick is to set the CLKPR register in software, before `setup()` is run:

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/cores/MCUdude_corefiles/wiring.c#L947-L951

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/boards.txt#L230

djwhouse, 2022-03-06:
Excellent! I understand better now. I assume when I compile the code with MiniCore set to 2MHz or 4MHz, the hex code that is generated will automatically include the CLKPR line into my code, so it runs at the correct frequency. I have seen some online information that suggested the fuses could be set to 4MHz or 2MHz, (https://www.ladyada.net/learn/avr/fuses.html, just one example), but never could find a fuse setting that allowed it. 

Is my assumption correct? 

Thank you for your quick reply!

Dave

MCUdude, 2022-03-06:
> I assume when I compile the code with MiniCore set to 2MHz or 4MHz, the hex code that is generated will automatically include the CLKPR line into my code

That's correct!

> I have seen some online information that suggested the fuses could be set to 4MHz or 2MHz, (https://www.ladyada.net/learn/avr/fuses.html, just one example), but never could find a fuse setting that allowed it.

This is only possible to do on "older" AVRs, such as ATmega8, ATmega16, and ATmega32.

djwhouse, 2022-03-06:
Thank you so much! Your Minicore was a great enhancement to my coding!!

closed 2022-03-06 by djwhouse

MCUdude, 2022-03-07:
You're welcome! 🙂

Floessie, 2022-10-24:
@MCUdude

Hans, I have a follow-up question concerning the 4 and 2 MHz settings: Suppose, I want to run at close to 1.8V. Then according to the datasheet, 4 MHz should be okay. You don't set the CKDIV8 for 4 MHz in your fuses, so the controller starts up at 8 MHz, which might not be stable for 1.8V.

Wouldn't it be better to boot with CKDIV8 set and speed up in the application rather than down?

Best,
Flössie