MCUdude/MiniCore #223: Low Fuse Question

vettett15, 2022-05-08:
Running the bootloader with a 2MHz external resonator but i'm curious how this is setup in the boards.txt file.  On my previous bootloader I would set the low fuse to 0xCB but it looks like it is being done differently here.  

In this minicore boards.txt file it looks like all of the external 328 low fuses are set the same:  0b1111{bootloader.cksel_bits}

Also, the clkpr line for each of them doesn't look to be filled out, thus i'm confused on how this is working?

Help,
Pete

MCUdude, 2022-05-08:
Hi Pete!

For external crystals/resonators, I'm using low fuse 0xF7. This works for all frequencies, and drives the crystal/resonator "as hard as it can", which means that it will consume slightly more power, but is very robust to noise and is rarely affected by capacitors that aren't properly matched to the crystal in use.

The clock prescaling line is only used when the internal  8 MHz oscillator is divided down to 4, 2 or 1 MHz. [Here](https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/boards.txt#L237) you can see the line where `build.clkpr` is used in boards.txt. the `build.clkpr` field is then added to the [`build.extra_flags`](https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/boards.txt#L55) field, and `build.extra_flags` are passed to the [avr-gcc build command](https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/platform.txt#L56-L63) as `{build.extra_flags}`. Now that `OSC_PRESCALER` may be defined, it can be utilized in the core files.

`OSC_PRESCALER` is used in wiring.c to set the approperiate CKLPR register value:

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/cores/MCUdude_corefiles/wiring.c#L947-L951

Hope this makes sense!

vettett15, 2022-05-08:
Hey thanks for the quick reply.  Yeah when I plug 0xf7 into the fuse calculator that makes sense.   Do you have any feel for how much more power it consumes in this mode?  I'm running the chip at 3.3v and my chip consumes 2 maybe 3mA 

For my edification, how does 0b1111{bootloader.cksel_bits} convert to 0xf7?

Thanks again
Pete

MCUdude, 2022-05-08:
> Do you have any feel for how much more power it consumes in this mode? I'm running the chip at 3.3v and my chip consumes 2 maybe 3mA

I haven't measured it, but I believe it is less than 1 mA.

> For my edification, how does 0b1111{bootloader.cksel_bits} convert to 0xf7?

Quite easy.

https://github.com/MCUdude/MiniCore/blob/18d2ac780376a985c0f343cabe95aee4d27dc3d7/avr/boards.txt#L88

```
0b1111{bootloader.cksel_bits} -> 0b11110111 -> 0xF7
```

vettett15, 2022-05-08:
Perfect, thanks.  I'm going to try and do some comparative current measurements 0xcb vs 0xf7. I'll report back if successful.  Worst case I assume I could just change the low fuse bit line under 2mhz external to 0xcb if 0xf7 didn't work for me

vettett15, 2022-05-13:
OK ran the test and it for me, custom board with 2mhz external resonator, the difference in current draw between the 0xf7 and 0xcb was just about 1mA as you predicted.  The 0xcb drew less current. 

Now I've got other problems.  I successfully burn the bootloader to my custom card using an arduino uno.  But when I disconnect the uno and go to upload a sketch to my custom card using my ftdi cable I get sync errors. 

I hooked the uno back up and I was able to upload the sketch using "arduino as isp" option but for whatever reason I can't do it using the "avrisp mkii" option with my ftdi cable.... gotta be something stupid

MCUdude, 2022-08-19:
> But when I disconnect the uno and go to upload a sketch to my custom card using my ftdi cable I get sync errors.

Which baud rate are you using? If you're using 9600 baud, which is reasonable, it's probably an auto-reset-related issue. Make sure you have a 100nF capacitor between the DTR line on the FTDI and the RESET pin.

MCUdude, 2022-10-03:
No answer from OP. Assuming issue is resolved.

closed 2022-10-03 by MCUdude