MCUdude/MiniCore #224: atmega328pb USART RX Start Interrupt

jaka87, 2022-05-11:
Hi.

If i understand correctly there is a possibility with the atmega328pb to wake up when i receive serial message. 
What i did is put 
`UCSR0D = 1 << RXCIE1 | 1 << RXSIE | 1 << SFDE;
`
in setup which (according to my understanding) enable RX flag. Then i tried to write ISR like this...
```
ISR(USART1_RX_vect)
{
// do something
}
```
which should trigger on new RX on Serial1 but i cannot compile it because i get 
```
/tmp/arduino_build_715208/../arduino_cache_885861/core/core_f3204311d3bdf6a1bbb83a41a5a6f099.a(HardwareSerial1.cpp.o): In function `__vector_28':
/home/jaka87/.arduino15/packages/MiniCore/hardware/avr/2.1.3/cores/MCUdude_corefiles/HardwareSerial1.cpp:48: multiple definition of `__vector_28'
/tmp/arduino_build_715208/sketch/UZH.ino.cpp.o:/home/jaka87/Arduino/UZH/UZH.ino:76: first defined here

```

I found very little information about this online so im hoping if someone here would know how to get RX wake up working. Help is greatly appreciated.
Regards, Jaka

MCUdude, 2022-05-11:
If you want to use the ISR directly you can't use `Serial1` _at all_. Replace all your `Serial1` related code and it should work.

jaka87, 2022-05-11:
Thank you for explanation. If i removed Serial1 it does compile but it doesn't work. I dont know if i should put different flags?

I tried 
```
ISR(USART1_RX_vect)
{
  DEBUGSERIAL.println("interrupt");
}
```
and i get no response from serial device

MCUdude, 2022-05-11:
Well, that's up to you to figure out 😉 For some reason, the ISR isn't firing. You're now running bare metal, so this isn't really MiniCore's fault. Just bear in mind that there are several USART1 registers that need to be initialized before you can use the serial port. Similar to what `Serial.begin()` does.

jaka87, 2022-09-22:
I finally figured it out in case someone else in the future will be having the same problem.
For Serial1 register should be set 
`UCSR1D = (1 << RXSIE) | (1 << SFDE)`

and then use ISR. In my case i only use it to wake up the device. There is no need to rewrite hardware serial.
```
ISR(USART1_START_vect){
countWake++;
}
```

Like [MCUdude](https://github.com/MCUdude) said its not really MiniCore related.

commit bf435d75 (2022-12-25): Fix incorrect TWBR register in twi1.c
Closes #224
closed 2022-09-22 by jaka87