MCUdude/MiniCore #225: Arduino Digital Pin 26?

JacksonAudio, 2022-05-19:
Are you able to access DP26 with arduino using the ATMEGA328PB? I've tried it but arduino doesn't seem to be able to reach it.

Any advise would be greatly appreciated!

MCUdude, 2022-05-20:
Are you sure there isn't something wrong with your hardware? I'm able to blink an LED using digital pin 26 (PE3) just fine with the following code:

``` cpp
void setup() {
  pinMode(26, OUTPUT);
}

void loop() {
  digitalWrite(26, HIGH);  
  delay(1000);    
  digitalWrite(26, LOW);
  delay(1000);
}
```

JacksonAudio, 2022-05-21:
That worked beautifully!
I have no idea what I did last night but it's working perfectly now. I feel baffled and embarrassed. :|

closed 2022-06-13 by MCUdude