MCUdude/MiniCore #226: 328pb freezing after seconds/minutes

fcig, 2022-06-09:
Hi.

I don't know exactly where the issue is but I would like a tip about this problem. I have a bunch of Arduino Nano clones with 328pb that I received from China. I was expecting the 328p but it's ok, I tried to figure out how to use them alongside with the 328p and MiniCore helped me a lot with this but recently I noticed a problem, they are randomly freezing with different sketches after seconds or minutes running.

I did some tests and found that setting them to use the internal oscillator at 8 MHz (burning the bootloader to set the fuses, and then uploading the sketch) makes them to run fine without any problem, but setting it back to external oscillator at 16 MHz (again, burning the bootloader to set the fuses, and then uploading the sketch) makes them unstable and they start freezing again.

I don't think the sketches are the issue because they run well on 328p at 16 MHz and with the 328pb at 8 MHz.

What could be the issue and what can I do to make them run well at 16 MHz?

MCUdude, 2022-06-09:
Hi!

First, can you post a high-res picture of your nano clone that came with a 328PB?

I'm pretty sure I know what the problem is. Unlike the 328p, the 328pb doesn't have a "full-swing oscillator option" in the fuse settings. This means that it will consume a tiny bit less power, but won't be able to drive the crystal/oscillator as "hard" as a 328p would. IMO it was a mistake by Microchip to remove this option because now the external oscillator has to be "matched" in order to not stop oscillating after a while.

I had to rework 100+ commercial-made PCBs I designed a few years ago since the manufacturer added a capacitor with built-in 33pF caps instead of 10pF caps. These boards were designed around the _very_ similar ATmega324pb and suddenly stopped working after a few weeks. To prevent this from happening again I switched to using a crystal + 2x 15pF capacitors, and never had any problems. 

I don't know what your board looks like, but you may have to replace the crystal/oscillator with a type that the 328pb is actually able to drive.  I can point you in the right direction as soon as I know the footprint of the current crystal/oscillator.

fcig, 2022-06-09:
Hi.

Sure, here are the pictures.

What you said makes sense since I already used a few nano clones with 328pb months ago and I didn't noticed this problem. Maybe they have the right oscillator and these I'm using now don't.

About changing the oscillator, thinking in buying them and changing (in other way, having more cost and time spent), I don't think it's worth for me changing it =/

In my case I think it's easier just use the nanos with 328p in projects that do need 16 MHz and reserve the nanos with 328pb to projects that have no problem to run at 8 MHz, but if you tell me that it would be possible to fix just placing a capacitor, certainly I would do it.

[image]
[image]

MCUdude, 2022-06-09:
Most of the stuff I design for commercial use runs at 3.3V because they are connected to an ADAU1701 audio DSP that runs at 3.3V. I'm running the AVRs at 8 MHz to keep them within spec.

I'm certain that 8 MHz is plenty for most applications. The most demanding application I'm running at 8 MHz is an ATmega1284P that acts as the brains in an audio amplifier. 
It communicates with a host over RS485, and has an OLED i2c screen with capacitive buttons one can use to navigate a menu where lots of DSP settings can be adjusted. The compiled program occupies about 117kB of flash, but the menus still feels snappy. And that's less than _half_ the speed of what the AVR could be running at.

fcig, 2022-06-09:
Thank you very much, helped a lot.
Yes, I agree, 8 MHz should do the job for most.

closed 2022-10-03 by MCUdude