MCUdude/MiniCore #228: ATmega328PB | Uploading a sketch via serial stops bootloader from working

defavltt, 2022-07-25:
I have made a couple of minimal configuration boards for ATmega328P with a 16mhz crystal, 15k reset pull-up and a filtering capacitor. Only when using the PB variant (with all the board settings being proper), when burning the bootloader, everything is fine, until when I upload an empty sketch for the first time after burning the bootloader, that works. Second time trying to upload the sketch and the ATmega328PB stops responding. 
MiniCore v2.1.3
I tried the following:
 - Checking if the crystal is oscillating
 - Burning a bootloader and trying to upload at different clocks: 16ext, 8int, 2int
 - Tricking it into using a 328P variant bootloader
 - Trying different bootloader serial speeds
 - Trying different CP2102 adapters (with a 200nf (works on every atmega) capacitor between MCU reset and adapter DTR pin)
 - It's not just this 328PB in particular (made a board from scratch for it, same problem)
 - Downloading the hex file from flash and comparing 1 - after burning the bootloader, 2 - after burning the bootloader and then uploading an empty sketch. The bootloader bytes at the end remain untouched, only the empty sketch bytes are added in the second one
 - Connections are correct since first sketch uploads fine

padnest, 2022-09-30:
Hi,
same issue solved in this way:

- burning 328PB bootloader on UART1 (not UART0) 
- FTDI programmer (5v) with 100nF cap from DTR to RESET

For some reasons bootloader on UART0 does not work! (avrdude: stk500_getsync(): not in sync...)

Ciao.

defavltt, 2022-10-18:
> Hi, same issue solved in this way:
> 
>     * burning 328PB bootloader on UART1 (not UART0)
> 
>     * FTDI programmer (5v) with 100nF cap from DTR to RESET
> 
> 
> For some reasons bootloader on UART0 does not work! (avrdude: stk500_getsync(): not in sync...)
> 
> Ciao.

Really appreciate it, got couple of 328PB's, don't want them collecting dust.

mcuee, 2022-12-21:
* https://github.com/Optiboot/optiboot/issues/360#issuecomment-1360827880

Somehow I could not get ATmega328PB Xplained Mini to work with either Optiboot hex file or MiniCore hex files.

However, the autobaud urboot firmware works fine.
https://github.com/stefanrueger/urboot/tree/main/bootloaders/atmega328pb/autobaud

In paticular, the following hex file is compatible with optiboot/arduino.
https://raw.githubusercontent.com/stefanrueger/urboot/main/bootloaders/atmega328pb/autobaud/urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex for flash
         with 470 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 42 pad bytes
avrdude: writing 470 bytes flash ...

Writing | ################################################## | 100% 0.13 s

avrdude: 470 bytes of flash written
avrdude: verifying flash memory against .\urboot_atmega328pb_autobaud_ee_lednop_fr_ce.hex

Reading | ################################################## | 100% 0.01 s

avrdude: 470 bytes of flash verified

avrdude done.  Thank you.
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14  -b 57600 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.31 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.23 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14  -b 115200 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.18 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.13 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -qqt
avrdude> dump lfuse
0000  e0                                                |.               |

avrdude> dump hfuse
0000  de                                                |.               |

avrdude> dump efuse
0000  fc                                                |.               |

avrdude> quit
avrdude>
```

0xCAFEAFFE, 2022-12-21:
The fuse bits above are E0 DE FC, which set the boot section size to 256 bytes, that can't be right?!

MCUdude, 2022-12-21:
> The fuse bits above are E0 DE FC, which set the boot section size to 256 bytes, that can't be right?!

It is correct. The boot size is set to 256 **words**, and one word is two bytes.

mcuee, 2022-12-21:
> Somehow I could not get ATmega328PB Xplained Mini to work with either Optiboot hex file or MiniCore hex files

Ah I did not notice the default optiboot hex file is 57600bps. Now it works. And the Optiboot hex file will hard code the signature the same as ATmega328P and not ATmega328PB.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -Pusb -e -Uefuse:w:0xFF:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xE0:m -q

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: reading input file 0xFF for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xFF
avrdude: 1 byte of efuse verified
avrdude: reading input file 0xDE for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xDE
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0xE0 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xE0
avrdude: 1 byte of lfuse verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\optiboot_xplained328pb.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14  -b 57600 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude error: expected signature for ATmega328PB is 1E 95 16
        double check chip or use -F to override this check

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14  -b 57600 -p m328p  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.31 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.23 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-12-21:
For the MiniCore bootloader files (Optiboot_flash), I can not get 115200bps to work, but 250000bps works fine.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c xplainedmini -p m328pb -U .\optiboot_flash_atmega328pb_UART0_250000_16000000L_B5.hex -qq && echo OK
OK
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM14 -b 250000 -p m328pb  -U .\Blink.ino.atmega328pb-xmini.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink.ino.atmega328pb-xmini.hex for flash
         with 928 bytes in 1 section within [0, 0x39f]
         using 8 pages and 96 pad bytes
avrdude: writing 928 bytes flash ...

Writing | ################################################## | 100% 0.11 s

avrdude: 928 bytes of flash written
avrdude: verifying flash memory against .\Blink.ino.atmega328pb-xmini.hex

Reading | ################################################## | 100% 0.07 s

avrdude: 928 bytes of flash verified

avrdude done.  Thank you.

```

mcuee, 2022-12-23:
I think this issue can be closed, as UART0 works well.

closed 2022-12-23 by MCUdude