MCUdude/MiniCore #229: atmega328 wont RESET

plusmartin, 2022-08-02:
[image]

I cannot understand how is it possible to NOT beingg able to RESET the atmega328 when pulling the RESET pin LOW.
After burning the code using the above settings it runs once and then it crashes and I cannot RESET the board unless I cycle power.

Isnt the RESET pin supposed to work always? Can it be somehow bypassed or ignored?

plusmartin, 2022-08-02:
If i press the RESET button while the code is still running then it does reset the sketch, but when it crashes it wont do anything.
I dont know if LTO should be enabled or disabled but it does the same issue on both cases.

plusmartin, 2022-08-02:
In other words, how can I enable the RESET pin so that it works again? thanks

MCUdude, 2022-08-19:
Are you able to burn the bootloader (set fuses) using an ISP programmer?

The Reset pin _can_ be configured as a GPIO, by clearing a bit in the low fuse memory. However, MiniCore does not disable the reset pin, but I don't know if you've been experimenting with the fuse bits or not...

You need a high-voltage programmer to recover a chip where the reset pin has been disabled, for instance an STK500 board.

plusmartin, 2022-09-05:
I have the STK500. Do I need atmel studio to program this fuse? do you know what value it requires?

Does this mean that every time I burn de MiniCore bootloader I am effectively bypassing the RESET pin and cannot set it back unless I burn the fuse manually again with atmel studio? This would require two programming steps, which is very unfortunate.

plusmartin, 2022-09-05:
Im sorry I misread your post. You state that MiniCore does NOT disable de reset pin, but for some reason my boards are not working using your bootloader.
Any suggestions on how to debug this? thanks

MCUdude, 2022-10-15:
No response from OP. Highly likely it's a hardware issue.

closed 2022-10-15 by MCUdude