MCUdude/MiniCore #232: Pololu ATmega328PB works / Minicore NOT

avandalen, 2022-09-08:
I use the Minicore ATmega328PB succesfully in some sketches.

But I have a strange compiler error with the Wattmeter-dongle sketch:
It works fine with the package: Pololu ATmega328PB installed
But with package Minicore ATmega328PB, I get this error:

`In file included from D:\DATA BACKUPPED\Documenten Albert\1 Solarbike development\Battery enz\0 Wattmeter dongle\Code\Wattmeter-dongle\Wattmeter-dongle\Wattmeter-dongle\Wattmeter-dongle.ino:56:0:
libLTC2944.h:35:3: error: extra qualification 'LTC2944_1024::' on member 'LTC2944_1024' [-fpermissive]
   LTC2944_1024::LTC2944_1024(float rSense);
   ^~~~~~~~~~~~
exit status 1
extra qualification 'LTC2944_1024::' on member 'LTC2944_1024' [-fpermissive]
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
[Wattmeter-dongle.zip](https://github.com/MCUdude/MiniCore/files/9525913/Wattmeter-dongle.zip)

MCUdude, 2022-10-03:
`-fpermissive` is the key here.

The Pololu 328PB core adds the -fpermissive flag, which turns certain errors into warnings. This is IMO a mistake, and will only hide code errors from the user since warnings are hidden by default in Arduino IDE. Tell the library author to fix their code, or modify the library code yourself to make it compile with MiniCore. I-ve decided to _not_ include the -fpermissive flag for all my "classic AVR" Arduino cores.

closed 2022-10-03 by MCUdude

avandalen, 2022-10-03:
Suddenly your minicore works again on my project, just like Pololu...
Both have no warnings or errors.