MCUdude/MiniCore #233: Cant upload with Arduino IDE 2.0.0

sl-maka, 2022-09-16:
Using MiniCore with Arduino 1.8.13 with no issues. Upgrade to 2.0.0 (and using all the same settings) still builds properly but the upload fails with the single error message "**A programmer is required to upload**". No further information... The connected programmer is a USBasp to a AtMega328p.  Running the 1.8.13 IDE again still works.

In both cases Im using MiniCore 2.1.3 on Debian GNU/Linux bookworm/sid x86_64

MCUdude, 2022-09-16:
If IDE 2.0 doesn't behave like 1.8.x, you should open an issue over at the [Arduino IDE 2.x repo](https://github.com/arduino/arduino-ide). I designed MiniCore to automatically upload using a programmer when "No Bootloader" was selected. It appears that IDE 2.0 doesn't do this. A quick fix is to select "Sketch > Upload using programmer", but I'd agree that the 1.8.x solution is better. 

You also need to select a programmer under "Tools > Programmer".

per1234, 2022-09-18:
Hi all.

>I designed MiniCore to automatically upload using a programmer when "No Bootloader" was selected. It appears that IDE 2.0 doesn't do this.

This deficiency in Arduino IDE 2.x is tracked here:

https://github.com/arduino/arduino-ide/issues/103

sl-maka, 2022-09-18:
Thanks for your responses. I'll monitor the issue there!

closed 2022-09-18 by sl-maka