MCUdude/MiniCore #234: 328PB - 8Mhz internal osc - serial programming works only with bootloader on UART1

padnest, 2022-09-30:
Hello,
I have this 328PB configuration on Arduino IDE (1.8.19):

- clock: internal 8 Mhz
- BOD: 2.7v
- variant: 328PB
- bootloader: yes (UART1 238PB only)

Serial programming via UART1 works without any problem.

If I try to burn the bootloader (UART0) serial programming does not work at all cause 'avrdude: stk500_getsync() attempt .. of 10: not in sync: resp=...'

What could be the reason?

Thanks.

MCUdude, 2022-09-30:
You need to re-burn the bootloader in order to switch to UART0.

padnest, 2022-10-02:
> You need to re-burn the bootloader in order to switch to UART0.

Reburning the bootloader to UART0 the serial programming on UART0 does not work...

MCUdude, 2022-10-03:
Sorry, but I can't reproduce your error. I can upload just fine using UART0 on my 328PB-based board. Are you sure nothing is connected to the UART0 pins on your 328PB? And are you sure your chip isn't broken? Can you read/write to UART0 when running a program on the 328PB?

padnest, 2022-10-03:
> Sorry, but I can't reproduce your error. I can upload just fine using UART0 on my 328PB-based board. 
Ok thank you for testing: are you using the same configuration? (clock internal 8 Mhz and BOD 2.7)

> Are you sure nothing is connected to the UART0 pins on your 328PB?
Yes, only the CP2102 UART-USB module (that is working fine on UART1).

> And are you sure your chip isn't broken? Can you read/write to UART0 when running a program on the 328PB?
UART0 seems right when used from the sketch program. The chip is brand new but let me check another chip.

I don't know if this other issue can indicate a bad chip: programming the bootloader with external 8 or 16 Mhz clock always fails cause: avrdude.exe: verification error, first mismatch at byte 0x7e00
The only working clock mode seems to be INTERNAL 8 Mhz.

On a 328P I've never had these issues with MiniCore (external clock 8/16 Mhz or UART0); on 328PB it's a nightmare! ;-))

MCUdude, 2022-10-03:
> cause: avrdude.exe: verification error, first mismatch at byte 0x7e00

This indicates that there's something wrong with the bootloader area of the chip. Can you turn on verbose upload in the Arduino IDE settings, and post _the entire_ output when trying to burn the bootloader on the 328PB? It might be a lock-bit fuse issue.

padnest, 2022-10-26:
Hi,
I tested another chip and all works fine on UART0 this time!.
Many thanks for your support.
Regards.

closed 2022-10-26 by padnest