MCUdude/MiniCore #238: Error when trying to load MiniCore in Arduino IDE 2.0.0

mg-man, 2022-10-08:
I'm trying to add MiniCore to Arduino IDE 2.0.0 and keep getting the following : 

Error: 2 UNKNOWN: Get "https://MCUdude.github.io/MiniCore/MiniCore-2.1.3.tar.bz2": read tcp [2a00:23c7:9219:500:3c55:e6bf:57ec:723b]:52356->[2606:50c0:8002::153]:443: wsarecv: An existing connection was forcibly closed by the remote host.

It looks like the .bz2 downloads (it's in my Downloads folder).  I've also tried extracting / putting the files in ~Arduino/Hardware, but that doesn't seem to work.  Any suggestions?

MCUdude, 2022-10-15:
@per1234 is this a known IDE 2.0 issue? The URL to the 2.1.3 bz2 file isn't dead...
I haven't had the time to properly test IDE 2.0 yet

mcuee, 2022-12-23:
It seems to be okay, tested with latest Arduino IDE 2.0.3.

```
Tool arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7 already installed
Tool arduino:avrdude@6.3.0-arduino18 already installed
Tool arduino:arduinoOTA@1.3.0 already installed
Downloading packages
MiniCore:avr@2.1.3
Replacing platform MiniCore:avr@2.1.2 with MiniCore:avr@2.1.3
Uninstalling MiniCore:avr@2.1.2
Platform MiniCore:avr@2.1.2 uninstalled
Uninstalling arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7, tool is no more required
Uninstalling arduino:avrdude@6.3.0-arduino18, tool is no more required
Uninstalling arduino:arduinoOTA@1.3.0, tool is no more required
Configuring platform.
Platform MiniCore:avr@2.1.3 installed

```

mcuee, 2022-12-23:
And I tested with burning the bootloader and upload the blink sketech on an ATmega328PB Uno Clone and MiniCore works fine.

closed 2022-12-23 by MCUdude