MCUdude/MiniCore #239: Compile fails when there is a space in the folder name

RaheelFarouk, 2022-10-13:
The compile fails when there is a space in any of the folder names. 
**For example:** 
C:\Program Files\code\code.ino <- this will fail to compile
C:\Program_Files\code\code.ino <- this will compile successfully 

**Error from Arduino IDE:**
'C:/Program' is not recognized as an internal or external command,
operable program or batch file.

exit status 1

Compilation error: exit status 1

harryberlin, 2022-10-13:
try "C:\Program Files\code\code.ino"

RaheelFarouk, 2022-10-13:
I am using the compile button on the Arduino IDE Version 2.0.0

Ziggy2013, 2022-10-15:
Arduino IDE 2.0 "improvements" at work.

MCUdude, 2022-10-15:
This is a known issue and was fixed in 71fa421. I'll have to fix a new boards manager release to fix this.

MCUdude, 2023-01-15:
MiniCore v2.2.0 is now available though boards manager!

closed 2023-01-15 by MCUdude