MCUdude/MiniCore #246: Expected signature for ATmega88 is 1E 93 0A

eddy-99, 2023-01-05:
I an having problems uploading a sketch in Arduino IDE 2.03.
I am using the ATtiny88 dev board.
I am uploading via USBasp. I have tried all the variants of the ATMega88 boards.
But still get this error.

>  Sketch uses 3406 bytes (44%) of program storage space. Maximum is 7680 bytes.
Global variables use 13 bytes (1%) of dynamic memory, leaving 1011 bytes for local variables. Maximum is 1024 bytes.
avrdude: Expected signature for ATmega88 is 1E 93 0A
         Double check chip, or use -F to override this check.
Failed programming: uploading error: exit status 1

MCUdude, 2023-01-05:
Can you turn on verbose upload in the IDE settings? I'd like to see which device signature the actual device has.

eddy-99, 2023-01-05:
Using board '88' from platform in folder: C:\Users\eddy_\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\eddy_\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3
Detecting libraries used...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o nul
Alternatives for Stepper.h: ***@***.***
ResolveLibrary(Stepper.h)
  -> candidates: ***@***.***
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o nul
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src\\Stepper.cpp" -o nul
Generating function prototypes...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp.o"
Compiling libraries...
Compiling library "Stepper"
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src\\Stepper.cpp" -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\librarie [... cut ...]
Using library Stepper at version 1.1.3 in folder: C:\Users\eddy_\AppData\Local\Arduino15\libraries\Stepper
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B/eddy_stepper.ino.elf"
Sketch uses 3406 bytes (44%) of program storage space. Maximum is 7680 bytes.
Global variables use 13 bytes (1%) of dynamic memory, leaving 1011 bytes for local variables. Maximum is 1024 bytes.

MCUdude, 2023-01-05:
Sorry, but that's the verbose compiler output, not the verbose Avrdude output

eddy-99, 2023-01-05:
is this the one.
Using board '88' from platform in folder: C:\Users\eddy_\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\eddy_\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.1.3
Detecting libraries used...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o nul
Alternatives for Stepper.h: ***@***.***
ResolveLibrary(Stepper.h)
  -> candidates: ***@***.***
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o nul
Using cached library dependencies for file: C:\Users\eddy_\AppData\Local\Arduino15\libraries\Stepper\src\Stepper.cpp
Generating function prototypes...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++17 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega88 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega88 -DARDUINO_ARCH_AVR "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\cores\\MCUdude_corefiles" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\2.1.3\\variants\\standard" "-IC:\\Users\\eddy_\\AppData\\Local\\Arduino15\\libraries\\Stepper\\src" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp" -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp.o"
Compiling libraries...
Compiling library "Stepper"
Using previously compiled file: C:\Users\eddy_\AppData\Local\Temp\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\libraries\Stepper\Stepper.cpp.o
Compiling core...
Using precompiled core: C:\Users\eddy_\AppData\Local\Temp\arduino-core-cache\core_68460d9feeccfeed7715e83d0583a3d8.a
Linking everything together...
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -w -Os -Wl,--gc-sections -mmcu=atmega88 -o "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B/eddy_stepper.ino.elf" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\sketch\\eddy_stepper.ino.cpp.o" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B\\libraries\\Stepper\\Stepper.cpp.o" "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B/..\\arduino-core-cache\\core_68460d9feeccfeed7715e83d0583a3d8.a" "-LC:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0CA572248AAD37AB97B" -lm
"C:\\Users\\eddy_\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\eddy_\\AppData\\Local\\Temp\\arduino-sketch-04AD634B18A8B0C [... cut ...]
         Programmer Type : usbasp
         Description     : USBasp, http://www.fischl.de/usbasp/

avrdude: auto set sck period (because given equals null)
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e9311
avrdude: Expected signature for ATmega88 is 1E 93 0A
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

Failed programming: uploading error: exit status 1

MCUdude, 2023-01-05:
It appears that you've actually connected to an **ATtiny**88. Atleast, the device signature matches:

from Avrdude.conf
```

#------------------------------------------------------------
# ATtiny88
#------------------------------------------------------------

part
    desc                   = "ATtiny88";
    id                     = "t88";
    prog_modes             = PM_SPM | PM_ISP | PM_HVPP | PM_debugWIRE;
    mcuid                  = 28;
    n_interrupts           = 20;
    stk500_devcode         = 0x73;
    chip_erase_delay       = 9000;
    pagel                  = 0xd7;
    bs2                    = 0xc2;
#   avr910_devcode         = 0x??;
    signature              = 0x1e 0x93 0x11;
```

eddy-99, 2023-01-05:
It is marked ATMEL

eddy-99, 2023-01-05:
it is marked 
T88 15MZ 1648 TH A809NA

MCUdude, 2023-01-05:
I realized now that you pretty much pointed out the issue in your first post.

> I am using the ATtiny88 dev board.
I am uploading via USBasp. I have tried all the variants of the ATMega88 boards.

> T88 15MZ 1648 TH A809NA

T88 means ATtiny88.

The ATmega88 and ATtiny88 are not the same chip, even though they are very similar. MiniCore does not support ATtiny88. Use [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) instead.

closed 2023-01-05 by MCUdude

eddy-99, 2023-01-05:
Sorry to trouble you and thanks for the quick reply.

Eddy