MCUdude/MiniCore #250: no usb detected after burning bootloader

luoGanda, 2023-01-26:
i don't know if this is miniCore issue but here it is(windows7-x64).
(i tried with 3 different cables, so this is not an issue, also with another computer running w7 - the same)

I bought nano(didn't know it didn't have a bootloader on) - with usb-b(micro plug).
Plugged in - no port com detected, tried installing ch341ser.exe, nothing.
Then i tried burning bootloader with miniCore via working ArduinoUno, did the process correctly in arduino ide - programming atmega386pb nano with uno.
Process went fine without any errors/warning, led blinked afterwards on both controllers.

Tried to unplug all, plugin in atmega386pb nano(now with bootloader), still nothing, no usb/serial port detected.
Any suggestion?

MCUdude, 2023-01-27:
It's not MiniCore related at all, but if the device doesnæt show up on your computer, and you've tried different USB cables, there is most likely a faulty board.

Either the CH340 chip on your board is defective, or there is a bad solder joint on the USB connector. It can either be a bridge or a pin on the connector that's not properly soldered

luoGanda, 2023-01-27:
the power light light up, nothing else.
I guess it's probably ch340 chip, since on chip - there is nothing physically written, blank chip.
This way, i can not even say it's ch340, altough i have tried two of them from same supplier from which images it's seen as ch340.
Ok, closing this.

mcuee, 2023-02-03:
This issue can be closed now.

closed 2023-02-15 by MCUdude