MCUdude/MiniCore #251: Flashing Bootloader to Arduino issue

unalearn, 2023-01-27:
Hello,

I keep getting this issue when trying to flash bootloader into a Microcontroller on breadboard using Arduino. When flashing other bootloaders (other than Minicore), it works well.

But when trying to flash bootloader using Minicore-2.2.1 & Minicore-2.2.0, I get into the issue as shown in screenshot.
[image]

unalearn, 2023-01-27:
Please suggest how to address this issue.

MCUdude, 2023-01-27:
Your installation is using Avrdude 6.3. You need to use Avrdude 7.1. If you install MiniCore using the boards manager, it will automatically download Avrdude 7.1

mcuee, 2023-02-03:
This is an installation issue so it can be closed.