MCUdude/MiniCore #252: Optiboot 8.0 32 pin cpu versus Minicore ATmega328 compilation

Erik84750, 2023-02-01:
Why can it be that a program written for ATmega328 would compile under Optiboot 8.0 boards definitions (32 pin cpu) would not compile when setting the board definitions for Minicore ATmega328P?

This is what is set under Additional Boards Manager:
https://github.com/Optiboot/optiboot/releases/download/v8.0/package_optiboot_optiboot-additional_index.json

A very small part of the verbose error output is here:

[code]
In file included from C:\Users\Erik Arckens\Google Drive\documents\Arduino\Nextion_DCCEX_Controller_V1.2.9.1\Nextion_DCCEX_Controller_V1.2.9.1.ino:142:0:
C:\Users\Erik Arckens\Google Drive\documents\Arduino\Nextion_DCCEX_Controller_V1.2.9.1\Nextion_DCCEX_Controller_V1.2.9.1.ino: In function 'void setup()':
DCCEX_Throttle.h:125:21: error: 'Serial1' was not declared in this scope
     #define nextion Serial1

[/code]

harryberlin, 2023-02-02:
328 does not have Serial1, only 328PB

Erik84750, 2023-02-02:
True, but this error is due to the wrong board definition being selected in the #if defined section.

So my question stands: what should be the correct syntax in my board definitions code for it to select the proper ATmega328P board when using MCUdude's board definitions?

Erik84750, 2023-02-02:
solved:
#elif defined (ARDUINO_AVR_ATmega328)
#define AVR_UNO

closed 2023-02-02 by Erik84750