MCUdude/MiniCore #253: sparkfun Tiny Usb ISP programmer not working after update

smdg49, 2023-02-08:
USBtinyISP (minicore) programmer fails to burn bootloaders in minicore versions greater than 2.1.3.

I get the folowing message from avrdude

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\******\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): found USBtinyISP, bus:device: 002:031
avrdude usbtiny_open() warning: cannot open USB device: Permission denied
avrdude usbtiny_open() error: cannot find USBtiny device (0x1781/0xc9f)
avrdude main() error: unable to open programmer usbtiny on port usb

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1

MCUdude, 2023-02-08:
This appears to be an Avrdude-related issue. MiniCore was upgraded to use Avrdude 7.1 after the 2.2.0 release.

What kind of machine and which OS are you running?

smdg49, 2023-02-08:
Windows 10
Arduino ide 2.03

mcuee, 2023-02-09:
> Windows 10 Arduino ide 2.03

Which driver are you using for your USBTinyISP? Are you using libusb0.sys or WinUSB driver?

smdg49, 2023-02-09:
libusb.sys and .dll

mcuee, 2023-02-10:
No issue with WinUSB driver.
```
avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): found USBtinyISP, bus:device: 003:020
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, https://learn.adafruit.com/usbtinyisp
avrdude pgm_default() error: programmer operation not supported

avrdude: using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
avrdude: erasing chip
avrdude: using SCK period of 10 usec
"C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.1-arduino.1/bin/avrdude" "-CC:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/avrdude.conf" -v -patmega328p -cusbtiny  "-Uflash:w:C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex:i" -Ulock:w:0xcf:m
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified
avrdude: reading input file 0b11111101 for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0b11111101
avrdude: 1 byte of efuse verified
avrdude: reading input file 0b11010110 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0b11010110
avrdude: 1 byte of hfuse verified
avrdude: reading input file 0b11110111 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
[... 51 lines cut ...]
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: using SCK period of 10 usec
avrdude: reading input file C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex for flash
         with 484 bytes in 2 sections within [0x7e00, 0x7fff]
         using 4 pages and 28 pad bytes
avrdude: writing 484 bytes flash ...

Writing | ################################################## | 100% 0.11s

avrdude: 484 bytes of flash written
avrdude: verifying flash memory against C:\Users\xfchen\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/bootloaders/optiboot_flash/bootloaders/atmega328p/16000000L/optiboot_flash_atmega328p_UART0_115200_16000000L_B5.hex

Reading | ################################################## | 100% 0.00s

avrdude: 484 bytes of flash verified
avrdude: reading input file 0xcf for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xcf
avrdude: 1 byte of lock verified

avrdude done.  Thank you.
```

mcuee, 2023-02-10:
> libusb.sys and .dll

@smdg49 

You can try the following.
1) Unplug and plug to see if it works
2) Switch to WinUSB driver if 1) does not work. You can use Zadig or the method here.
https://github.com/avrdudes/avrdude/wiki/Troubleshooting-USB-devices-in-Windows#install-the-winusb-driver-for-a-usb-device

smdg49, 2023-02-12:
WinUSB works but only once.
after that I get

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is
C:\Users\Shmuel\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbtiny
         Setting bit clk period        : 32.0
avrdude: usbdev_open(): found USBtinyISP, bus:device: 002:013
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page
                    Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size
#Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ----
------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4
   0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128
 256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1
   0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1
   0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1
   0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1
   0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1
   0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1
   0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer,
https://learn.adafruit.com/usbtinyisp
avrdude pgm_default() error: programmer operation not supported

avrdude: setting SCK period to 32 usec
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000 (retrying)
avrdude: device signature = 0x000000
avrdude main() error: Yikes!  Invalid device signature.
avrdude main() error: expected signature for ATmega328P is 1E 95 0F
        Double check connections and try again, or use -F to override
        this check.

avrdude done.  Thank you.

Error while burning bootloader.

I bricked a few chips like this all were smd  atmega328p

Arduino as ISP does not work either in the new minicore version.

mcuee, 2023-02-12:
> WinUSB works but only once.
> I bricked a few chips like this all were smd  atmega328p

That is strange. This should not happen. Are you saying the old avrdude 6.3 works for your setup yet the new avrdude 7.1 does not work?

> Arduino as ISP does not work either in the new minicore version.

Again, does the old avrdude 6.3 from Arduino work for you setup?

smdg49, 2023-02-12:
All I know is that thanks to minicore I have been using this method
without any problems. Until i upgraded.

Will try everything tomorrow in detail.

smdg49, 2023-02-12:
Minicore 2.1.3  libusb
*************************************
Nano
***********************************
"C:\Users\Shmuel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude"
"-CC:\Users\Shmuel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"
-v -patmega328p -cusbtiny  -e -Ulock:w:0x3F:m -Uefuse:w:0xFD:m
-Uhfuse:w:0xDA:m -Ulfuse:w:0xFF:m

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is
"C:\Users\Shmuel\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device:
bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page
          Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW
 MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------
----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0
 3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256
 4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0
 4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0
0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0
0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer,
https://learn.adafruit.com/usbtinyisp
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

[... 862 lines cut ...]
avrdude: 32768 bytes of flash verified
avrdude: reading input file "0x0F"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0F:
avrdude: load data lock data from input file 0x0F:
avrdude: input file 0x0F contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

So nano only works with libusb
new minicore only works with winusb driver

I guess I bricked all of my atmega328p(s) when I was trying to troubleshoot
and accidently set it to 16mhz external. will have to make an smd board
with a clock to restore them.

mcuee, 2023-02-13:
> So nano only works with libusb
> new minicore only works with winusb driver

It is expected that avrdude 6.3 from Arduino only works with libusb0.sys (libusb-win32 device driver), as it uses dynamic link to to libusb0.dll (libusb-win32 dynamic library).

As for avrdude 7.1 from Arduino, it is recommended to use WinUSB driver, as it linked to libusb-compat-0.1 library (static link) which is on top of libusb-1.0 library. By right, libusb0.sys should also work but there are problems with libusb0.sys support for libusb-1.0.

If you use the official avrdude 7.1 Windows binary from avrdude project, it should work with both libusb0.sys driver or WinUSB driver, as it is using static link to avrdude-libusb.
https://github.com/avrdudes/avrdude/releases/tag/v7.1

Reference:
* https://github.com/arduino/avrdude-packing/issues/15
https://github.com/libusb/libusb/wiki/Windows#user-content-Driver_Installation
https://github.com/avrdudes/avrdude/wiki/The-story-on-libusb-for-Windows
https://github.com/avrdudes/avrdude/wiki/Troubleshooting-USB-devices-in-Windows

mcuee, 2023-02-13:
@smdg49 

If you want to further debug the issue (not with avrdude, but with libusb project), you can add more info here.
* https://github.com/libusb/libusb/issues/1226

smdg49, 2023-02-13:
>    - libusb/libusb#1226 <https://github.com/libusb/libusb/issues/1226>

k thanks

mcuee, 2023-02-13:
@smdg49 

What about "Arduino as ISP"? It does not use libusb at all so the issue is different.

smdg49, 2023-02-13:
Did not look into that yet. Rarely use it.

mcuee, 2023-02-13:
> Did not look into that yet. Rarely use it.

No problem.

BTW, did you encounter the issue every time or not? Or is it occasionally? When you encounter the issue, does it help if you unplug and plug it back? 

```
"C:\Users\**********\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.1-arduino.1/bin/avrdude" 
"-CC:\Users\*************\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/avrdude.conf" 
-v -patmega328p -cusbtiny  -e -Ulock:w:0xff:m -Uefuse:w:0b11111101:m 
-Uhfuse:w:0b11010110:m -Ulfuse:w:0b11110111:m

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\********\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): found USBtinyISP, bus:device: 002:031
avrdude usbtiny_open() warning: cannot open USB device: Permission denied
avrdude usbtiny_open() error: cannot find USBtiny device (0x1781/0xc9f)
avrdude main() error: unable to open programmer usbtiny on port usb

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1

```

smdg49, 2023-02-13:
Every time

mcuee, 2023-02-13:
> Every time

Please help to post the debug log from arvdude first by running the following in Windows command line. 

```
"C:\Users\**********\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.1-arduino.1/bin/avrdude" 
"-CC:\Users\*************\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.1/avrdude.conf" 
-vvvv -patmega328p -cusbtiny 
```

Then do the same thing but with Environment Variable `LIBUSB_DEBUG` set to `4`.

smdg49, 2023-02-13:
Arduino IDE is still using avrdude 6.3.x

mcuee, 2023-02-13:
> Arduino IDE is still using avrdude 6.3.x

Sorry but I do not understand this sentence. Arduino IDE with MiniCore 2.2.1 will use avrdude 7.1 version from Arduino. 

To troubleshoot the issue, you do not need to use Arduino IDE at all. Rather you just use the command line to see what is the problem.

To make it easier to you, maybe you can download the binary here and try it. It is the same as the version shipped with MiniCore 2.2.1.
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1
https://github.com/arduino/avrdude-packing/releases/download/7.1-arduino.1/avrdude_7.1-arduino.1_Windows_32bit.tar.gz

mcuee, 2023-02-14:
@smdg49 

It is okay if you find it troublesome to troubleshoot the issue as it is not really related to MiniCore or even avrdude 7.1, rather it is related to libusb-1.0 Windows support for libusb0.sys.

As mentioned,   you should use WinUSB driver for avrdude 7.0/7.1 under Windows in general and not libusb0.sys.

smdg49, 2023-02-14:
Thanks.
I also switched out avrdude arduino boards to 7.0.x. everything seems to
work fine now with winusb driver

mcuee, 2023-02-14:
> Thanks.
> I also switched out avrdude arduino boards to 7.0.x. everything seems to work fine now with winusb driver.

In that case, is it okay that you can close this issue? Thanks.

closed 2023-02-15 by smdg49