MCUdude/MiniCore #255: avrisp mkII not communicating with avrdude in V2.2.1

spelect, 2023-04-08:
Hi Hans
Many thanks for your great MiniCore 👍 

I recently upgraded to minicore to 2.2.1 and found the included new version of avrdude 7.1 does not communicate with my avrisp MKII.  I rolled back to 2.1.3 and it all works fine (avrdude 6.3)
Regards Scott

MCUdude, 2023-04-08:
Hi Scott!

You're absolutely right, there was a bun in the programmers.txt file that prevented the avrispmkii programmer option from working. I fixed the issue a while ago but haven't pushed a new boards manager release yet. If you want to use the latest version, you can modify the programmers.txt file, and it will work with the avrispmkii again.

```diff
 avrispmkii.name=AVRISP mkII
 avrispmkii.communication=usb
-avrispmkii.protocol=stk500v2
-avrispmkii.program.protocol=stk500v2
+avrispmkii.protocol=avrispmkii
+avrispmkii.program.protocol=avrispmkii
 avrispmkii.program.tool=avrdude
 avrispmkii.program.extra_params=
```

spelect, 2023-04-08:
Many thanks for your prompt reply Hans :)
Will have go

closed 2023-04-08 by spelect