MCUdude/MiniCore #261: Can not use atmega8 with external crystal

rayle1360, 2023-07-15:
I am trying to use atmega8 with external crystal but with no luck. internal oscillator works fine and I can upload sketch without any problem. but when I burn bootloader with external 16mhz crystal, I get "avrdude error: programmer is not responding". the crystal is fine because when I use menu sketch>upload using programmer then it works. but I want to use it like a real Arduino to upload sketch via serial. any sugestions??

MCUdude, 2023-07-15:
In order to use an external oscillator, you need to set the correct fuse bits. You do this by selecting the appropriate settings under the tools many and click "Burn bootloader".

Can you post the complete error message you get when you're getting the error message form Avrdude?

rayle1360, 2023-07-15:
thanks for you attention
I do it exactly as you said, and after "Burn bootloader" its fusebits are set correctly because i can not re Burn bootloader without a crystal. but for some reason it does not upload sketch via UART. the complete error is "avrdude error: programmer is not responding"

rayle1360, 2023-07-15:
actually after testing for many times I managed to successfully upload a simple sketch. but but a successful upload happened just 2 times out of, I don't know maybe more that 50 times. and still can not upload again I get error "avrdude error: programmer is out of sync"

rayle1360, 2023-07-16:
I'm sorry, it seems that the problem was with my USBasp programmer. I was using it to burn the bootloader and it was causing the issue. I used another Arduino as an ISP, and now it works like a charm. Thank you.

rayle1360, 2023-07-29:
after lots of testing it seems that it does not want to work. the previous time was a one time that worked. after lots of testing it refuse to work with 16mhz external crystal. but with internal crystal it works well. I double checked the fuses and all are correct. any idea why it is not working. after burning boot loader when I try to upload a simple blink sketch it says "avrdude error: programmer is not responding". that's it

Creo2005, 2023-12-10:
> after lots of testing it seems that it does not want to work. the previous time was a one time that worked. after lots of testing it refuse to work with 16mhz external crystal. but with internal [...]

Please let me know which urboot file i need to take for flash Atmel8. And what should the fuse bit settings be to use the internal crystal (1 or 8 MHz). Because I set the fuse incorrectly and now uasasp programmer sees the atmega8 only with external cristal. I want to remove the external crystal. Here's the project I made https://github.com/jpraus/arduinoflake, but now I can't set the fuse correctly to work without external cristal and i whant flash the urboot  and them flash firmware via usb from Arduino IDE. Thanks.

rayle1360, 2023-12-13:
> > after lots of testing it seems that it does not want to work. the previous time was a one time that worked. after lots of testing it refuse to work with 16mhz external crystal. but with internal crystal it works well. I double checked the fuses and all are correct. any idea why it is not working. after burning boot loader when I try to upload a simple blink sketch it says "avrdude error: programmer is not responding". that's it
> 
> Please let me know which urboot file i need to take for flash Atmel8. And what should the fuse bit settings be to use the internal crystal (1 or 8 MHz). Because I set the fuse incorrectly and now usbasp programmer sees the atmega8 only with external cristal. I want to remove the external crystal. Here's the project I made https://github.com/jpraus/arduinoflake, but now I can't set the fuse correctly to work without external cristal and i whant flash the urboot and them flash firmware via usb from Arduino IDE. Thanks.

you don't need to set any fuse, just select atmega8 and then select the click 8mhz and finally click on burn bootloader. that is it

closed 2024-10-10 by MCUdude