MCUdude/MiniCore #262: Minicore 328pb   and    DS1307RTC

Zlodeyz, 2023-07-20:
При использовании библиотеки DS1307RTC в демо скейтчи  Readtest  arduino nano не отрабатывает setup (зависает)
Вся проблема в очередности подключения библиотеки.

Так НЕ работает
#include <Wire.h>
#include <DS1307RTC.h> 

Так работает.
#include <DS1307RTC.h> 
#include <Wire.h>   

В чем проблема ?

MCUdude, 2023-07-20:
You're using this library, right? https://github.com/PaulStoffregen/DS1307RTC

It includes Wire.h in its .cpp file, so there's no need to include it in your sketch. I have no idea why it freezes. Ask the library author.

Zlodeyz, 2023-07-20:
You're using this library, right? https://github.com/PaulStoffregen/DS1307RTC
yes .
Спасибо

Zlodeyz, 2023-07-20:
Если выбрать стандартную плату arduino nano то таких проблем нет.