MCUdude/MiniCore #265: Bootloaders burned in products?

Fkulo, 2023-08-20:
Hi MCUdude!
Thank you for all of your amazing cores that let us program chips properly!I have a question regarding bootloaders burned in commercially sold products.I have read the GNU licenses that your MiniCore software complies by, but I am sort of lost as to how burned bootloaders in microcontrollers inside of sold products would comply.I am planning on creating a commercially available product that serves educational purposes.It would have an AtMega328PB inside of it,that would have the bootloader burned from the MiniCore, and after someone would buy the product, there would be referal links to your MiniCore github page, where they would be able to install the core to thier arduino ide and then program the chips themselves.I also wanted to ask for your permission to do so, after all you are the creator of this software.If you would like to know more about the product and what it does,I do have some prototypes I could show.
Thank you again for your awesome sauce stuff, and in advance for your reply.

MCUdude, 2023-08-20:
Hi @Fkulo!
Great to hear you find MiniCore useful, and that you want to use it in a commercial product!

I'm totally fine with anyone that wants to use MiniCore in commercial products. The license (LGPL 2.1) was chosen since this project is based on the original Arduino AVR source files. And as long as you comply with the license, I'm fine with whatever 🙂 

The same goes for Optiboot. Arduino has bundled the bootloader for more than a decade and it has never been an issue with this AFAIK. However, if you do any modifications to the source code (MiniCore or Optiboot), you'll have to make them open source (I think). But as long as you point to the MiniCore Github page, I'm sure you're OK.

I don't make any money on the software projects I host on Github, so I'm just happy that more users get to use them.

Regarding the 328PB design. Just bear in mind that this chip doesn't have the "Full swing oscillator" fuse option, so the external oscillator will be way more susceptible to noise and matched crystal/load capacitance than the more common ATMega328P. I've designed a commercial product around the very similar ATmega324PB, and I ended up using an external clock that feeds a square wave signal straight into the XTAL1 pin.

Fkulo, 2023-08-20:
@MCUdude 
Thank you very much!
Also thanks for the information regarding the PB chips, the chip in the product can be whatever as long as it has 13 digital output pins to turn on LED-s with and it can be programmed trough the arduino ide and the regular uart interface.

The chip will have a switching converter about 4-5cms next to it that could mess things up in this case so thanks for the heads up!

MCUdude, 2023-08-20:
> The chip will have a switching converter about 4-5cms next to it that could mess things up in this case so thanks for the heads up!

The main problem I was facing was too much capacitance on the XTAL lines, and the chip would just stop executing instructions after a few days or weeks. 

Since the 328PB are SMD only, you might as well add a few more SMD parts while you're at it. An accurate 16 MHz clock (or any clock you'd prefer) costs ~0.1USD, so not much more than a crystal + load capacitors.

Fkulo, 2023-08-20:
Problems that dont show themselves for days and then suddenly hit are the worst.
I would'nt be against going a 16mhz clock rout,it would mean resonator instead of crystal right?I only recently got into MCUs and digital tech., was a hardcore analog guy for a while until I realized how cool digital actually is.
Also would using the internal oscillator be a problem?The chip only has to execute simple digitalWrite HIGH/LOW commands so it wouldnt really need to be very accurate in terms of timing.Also thought this way I could reduce the capacitance on the XTAL pins, they would only have a ground plane under or right next to them the size of the actual pad of the chip this way.But I also read that internal 8mhz crystals cause problems with programming.

MCUdude, 2023-08-20:
> it would mean resonator instead of crystal right?

You have crystals (that require external load capacitors), resonators (that usually have capacitors built-in), and oscillators that provide a low-impedance clock output that's fed straight into the XTAL1 pin.

I actually had issues with a Murata branded resonator where the assembled house had replaced one that should have had 10pF load capacitance with one that had 30pF (or was or 33?). 

> Also would using the internal oscillator be a problem?

Maybe. Maybe not. For IO toggling, it's not a problem, but I have experienced issues in the past where the internal oscillator couldn't speak UART because its internal clock was too slow and needed to be calibrated. This is absolutely possible for a production board that's going to be flashed once, but I wouldn't recommend it on a development board, since the user program has to deal with the new clock calibration value (OSCCAL) value on every boot. And the bootloader would also have to deal with it, something that Optiboot currently doesn't do. So a 10-cent external crystal/resonator/oscillator would save you a lot of trouble IMO.

Fkulo, 2023-08-20:
Thank you again for all of this information!
I am sorry for being difficult and asking all of these questions.
So to my understanding I can solve the problem by adding an oscillator that you mentioned instead of crystals/resonators to the existing 328PB,or use a 328p or equivalent  MCU that doesn'nt have a problem with crystals/resonators.

MCUdude, 2023-08-20:
> I am sorry for being difficult and asking all of these questions.

No worries! 

> So to my understanding I can solve the problem by adding an oscillator that you mentioned instead of crystals/resonators to the existing 328PB, or use a 328p or equivalent MCU that doesn't have a problem with crystals/resonators.

Correct. You _can_ absolutely use a standard crystal or resonator with the 328PB, but keep in mind that it's way, way more picky when it comes to the load capacitance (to ground), so keep the XTAL1 and XTAL2 traces as short as possible, and read the 328PB datasheet to make sure you're getting it just right. And as I mentioned, but just a few more cents you can instead use an external oscillator/clock that solves all these potential issues.

However, the 328P can pretty much drive anything, and it's very forgiving when it comes to load capacitance. The 328P is also more common in the "Arduino world", so more 3rd party libraries will be supported. However, it's very simple to make a library support the 328PB, but it will require some modifications to the library source code. It's your choice 😉

Fkulo, 2023-08-21:
Thank you very much!
I think I will go with a 328p.

MCUdude, 2023-08-21:
Good choice! That's what I'd recommend if you don't absolutely need the additional feature the 328PB provides.

closed 2023-08-21 by MCUdude

Fkulo, 2023-08-21:
Hi @MCUdude !
First of all thank you again for your help with the previous matters.
Secondly I have looked around a bit and I would like to ask for your opinion regarding MCUs if I may.
I mentioned that I did'nt need that much processing power when it comes to this project of mine, and found chips that the internet and datasheets tell me are sort of similar to the 328p, namely the AtMega48PA-AU.Do you have some experience with this chip?

MCUdude, 2023-08-21:
Pretty much _all_ AVRs have the same processing power. The thing that differs is the memory, number of pins, and peripherals. If your program can fit on an ATmega48(PA) you can absolutely use this. But keep in mind that MiniCore currently doesn't offer a bootloader for the ATmega48, so your'll have to use an ISP programmer. The other chips (ATmega8/A, ATmega88/168/328A/P/PA/PB) already has bootloader support through MiniCore.

This will soon change when I release the 3.0.0 version of this core, where I replace Optiboot with the superior [Urboot](https://github.com/stefanrueger/urboot) bootloader. I'm currently working on the transition for [MightyCore](https://github.com/MCUdude/MightyCore/tree/urboot), but it will take several months before I have time to start working on the MiniCore transition.

Fkulo, 2023-08-21:
Thank you very much!

Fkulo, 2023-08-22:
Now I see that you mentioned ATmega8/A for chips that have MiniCore support.
I apologize for being a complete noob and bombarding you with questions.