MCUdude/MiniCore #266: Debug support

felias-fogg, 2023-09-12:
Hi Hans,

I worked on dw-link, the debugWIRE debugging solution, and think I have simplified the setup quite a lot: https://github.com/felias-fogg/dw-link/blob/master/docs/quickstart.md. I took your MiniCore files and extended it by an option to generate binaries that are "debugging-friendly". In the course of doing so, I noticed that IDE 2.X has "changed the rules" by not defining the variable {sketch_path}, which led to an error under macOS. This led me to provide the script` savecp.sh`, which copies only if the destination is not "{sketch_path}". I can provide a pull request (together with a Python script that adds the necessary definitions in the future).  

Best,
Bernhard

P.S.: If you send me your postal address to nebel@hinterm-ziel.de, I will send you an assembled dw-link probe shield ;-).

MCUdude, 2023-09-14:
Thanks for the reach-out! I had a brief look at your work, and it looks very comprehensive and well documentated.
I'd love to test your dw-link probe on some actual hardware, so expect an email from me soon.
I realize that the debugger is meant to be used with a GDB GUI or PlatformIO, but does it work with Arduino IDE 2.0 as well?
I just read about this project where Arduino IDE 2.0 is used for AVR debugging: https://github.com/DeqingSun/unoDebugTestPackage

I'm about to merge the [urboot branch](https://github.com/MCUdude/MiniCore/tree/urboot) into master and release MiniCore v3.0.0, so PRs should be submitted to the urboot branch.

Have you looked into how one could use _official_ AVR debuggers as well with PlatformIO (and Arduino IDE 2.0 perhaps)? Ever since I got involved in the [Avrdude](https://github.com/avrdudes/avrdude) development (new developers welcome to engage!), I've acquired lots of different debuggers I can't really use for debugging without closed-source Microchip software. I know Avarice exists, but I don't know how up-to-date it is, nor how to utilize it.

felias-fogg, 2023-09-14:
> I realize that the debugger is meant to be used with a GDB GUI or PlatformIO, but does it work with Arduino IDE 2.0 as well?
> I just read about this project where Arduino IDE 2.0 is used for AVR debugging: https://github.com/DeqingSun/unoDebugTestPackage

I was not successful yet in replicating what DeqingSun did. He told me that there were changes in the IDE that invalidated some of his "hacks". Well, the main problem here is that Arduino does not publish their interface to debugging and that you have to guess and hack, e.g., you have to rename "avr-gdb" to "arm-none-eabi-gdb". So, if I come up with a hacked solution, then tomorrow this will not work anymore, which is very unpleasant.  

> Have you looked into how one could use _official_ AVR debuggers as well with PlatformIO (and Arduino IDE 2.0 perhaps)? Ever since I got involved in the [Avrdude](https://github.com/avrdudes/avrdude) [...]

I certainly did! I myself also own an Atmel ICE, and only could get it to work on a Mac with the MPLAB X IDE, which is not a system I love to work with. I never got AVaRICE to work on a Mac and did not find enough info on the web to help me. Well, if some of these things had worked out for me, I would not have started to work on dw-link.

per1234, 2023-09-27:
> IDE 2.X has "changed the rules" by not defining the variable {sketch_path}

This bug is now tracked by the Arduino CLI developers at https://github.com/arduino/arduino-cli/issues/2340. There is no need to comment there. I'm only adding this comment to notify affected users that there is a formal bug report about it in the appropriate issue tracker.

MCUdude, 2023-10-21:
@felias-fogg I've been quite busy recently, and I can't promise any progress when it comes to the MiniCore integration just yet. However, I think the project is really impressive, and I will see what I can do when I have more time.

felias-fogg, 2023-10-21:
Hi Hans,

no hurry. I am currently busy with other things as well ;-)

Best.
Bernhard

felias-fogg, 2024-12-20:
Hi Hans,

I picked up on the "left-over" problems and now have a working debugger version for MiniCore in the Arduino IDE 2.3.4. I used DeqingSun's board manager files and after 3 days of trial and error, it appears to work now. The AVRs pretend to be debugged by openocd ;-) and for this reason, it is more or less compatible with the cortex-debug extension of VSCode, which is used by the Arduino IDE 2.X.

What I would like to do now is to produce is a board manager file using your most up-to-date version as a base. So, my question is: Do you have any scripts for producing it or pointers to documents about what one has to do? 

Cheers,
Bernhard

felias-fogg, 2024-12-23:
Hi,

Just found the scripts. I hope to have a release ready by the end of this year.

Merry Christmas and a Happy New Year,
Bernhard

closed 2025-05-01 by felias-fogg