MCUdude/MiniCore #271: Weird issues with ArduinoISP when running from MiniCore

vanyasem, 2023-10-21:
After flashing MiniCore as the bootloader on Uno and two Nanos the ArduinoISP sketch stops working on them

However, ArduinoISP sketch works fine on the default Arduino bootloader

I have even tried using https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/main/ArduinoISP/ to see if it would make any difference. It did not. I have tried messing with the settings to no success.

I have tried all possible combinations. ArduinoISP does not want to work on MiniCore devices

I have tried:
Target Nano (minicore) - Destination Nano (default);
Target Nano (minicore) - Destination Uno (default);
Target Uno (minicore) - Destination Leonardo (default);

I can't come with a reasonable explanation for this, and I have failed to troubleshoot the issue myself. I'd appreciate any feedback on this

`avrdude` log (ArduinoISP sketch running on MiniCore Uno trying to flash leonardo)
```
"$HOME/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude" "-C$HOME/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf" -v -patmega32u4 -cstk500v1 -P/dev/cu.usbserial-14220 -e -Ulock:w:0x3F:m -Uefuse:w:0xcb:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/vanyasem/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/Users/vanyasem/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-14220
         Using Programmer              : stk500v1
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xa0
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x77
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xa0
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x77
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xa0
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x77
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xa0
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x77
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xa0
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x77
```

ArduinoISP sketch running on the same Uno with stock bootloader with the same wiring successfully flashes this exact leonardo

MCUdude, 2023-10-21:
Which version of MiniCore have you installed, and have you installed it using the boards manager?
I've just released v3.0.0, which replaces Optiboot in favor of the superior [Urboot](https://github.com/stefanrueger/urboot) bootloader.

vanyasem, 2023-10-21:
Oops, forgot to mention the most important thing! I am using MiniCore **v3.0.0** @MCUdude

MCUdude, 2023-10-21:
> I am using MiniCore v3.0.0

Thanks! Did you install the bootloader on the UNO using MiniCore v3.0.0, or did it already have a bootloader (likely Optiboot)?

vanyasem, 2023-10-21:
@MCUdude 
I have never had any custom bootloader before. MiniCore v3.0.0 was the first custom bootloader that I've installed

I have also tried reverting to the original bootloader (to test if ArduinoISP works - it does), and then flashing MiniCore v3.0.0 again (and ArduinoISP broke)

MCUdude, 2023-10-21:
My time is quite limited, but I'll have a closer look at the issue within a few days.
Maybe @stefanrueger (who created the Urboot bootloader) knows why the Urboot bootloader breaks *Arduino as ISP*?

stefanrueger, 2023-10-21:
Urboot bootloaders need
 - `avrdude -c urclock` for programming, which is only available from AVRDUDE v7.0. (v6.3 does not work)
 - Potentially [different fuse settings](https://github.com/stefanrueger/urboot#Usage) than optiboot

So, try to figure out which urboot bootloader you have and see whether the fuse settings are correct

vanyasem, 2023-10-22:
@MCUdude what are the flags used for `urboot` in MiniCore?

[image]

vanyasem, 2023-10-22:
I'll compare the fuse settings between stock bootloader and MiniCore and will get back with results

MCUdude, 2023-10-22:
> what are the flags used for urboot in MiniCore?

the Urboot bootloader is built with these flags/features:

```
avrdude -c urclock -P /dev/cu.usbmodem14101 -p atmega328p -x showversion
avrdude: AVR device initialized and ready to accept instructions
u7.7 weu-jPrac
```

I'm not able to reproduce the issue you're seeing though.
I uploaded the ArduinoISP sketch available under `File > Examples > 11. ArduinoISP` to a genuine Arduino UNO running Urboot. I wired up the UNO to a target board, in my case an Atmega1284P, and connected a capacitor between GND and the RESET pin on the UNO. No issues here...

```
/Users/hans/Library/Arduino15/packages/MightyCore/tools/avrdude/7.2-arduino.1/bin/avrdude -C/Users/hans/Library/Arduino15/packages/MightyCore/hardware/avr/3.0.0/avrdude.conf -v -patmega1284p -cstk500v1 -P/dev/cu.usbmodem14101 -b19200 -Ueeprom:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.eep:i -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.hex:i 

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Library/Arduino15/packages/MightyCore/hardware/avr/3.0.0/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbmodem14101
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash                  65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  9000  9000 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9705 (probably m1284p)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U eeprom:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.eep:i
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.eep for eeprom
         with 0 bytes in 0 sections within [0, -1]
         using 0 pages and 0 pad bytes
avrdude: writing 0 bytes eeprom ...
Writing | ################################################## | 100% 0.00s
avrdude: 0 bytes of eeprom written
avrdude: verifying eeprom memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.eep
Reading | ################################################## | 100% 0.00s
avrdude: 0 bytes of eeprom verified

avrdude: processing -U flash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.hex:i
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.hex for flash
         with 1012 bytes in 1 section within [0, 0x3f3]
         using 4 pages and 12 pad bytes
avrdude: writing 1012 bytes flash ...
Writing | ################################################## | 100% 1.11s
avrdude: 1012 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_652192/Blink.ino.hex
Reading | ################################################## | 100% 0.58s
avrdude: 1012 bytes of flash verified

avrdude done.  Thank you.

```

[image]