MCUdude/MiniCore #272: urboot bootloaders in MiniCore only for 1 frequency external crystal

hwvern, 2023-11-04:
I can't find the different bootloaders for 16MHz en 20MHz external crystals in the packages/MiniCore/hardware/avr/3.0.0/bootloaders/urboot/atmega328p/ folders. On Github of StefanRueger they can be found both, and they seem to differ.

MCUdude, 2023-11-04:
If you download the entire [urboot.hex](https://github.com/stefanrueger/urboot.hex) repo and unzip it, it occupies over 70 GB with more than 5 million files. That's _way_ too much to bundle with MiniCore, so I've only added the ones this core will use though Arduino, and in the future, PlatformIO.

The reason why you can't find different bootloaders for 16 and 20 MHz is because there isn't any! The bootloader is utilizing autobaud, which automatically adjusts to the host (Avrdude) baudrate. When you have a bootloader that can do this, there's no need to have separate bootloaders for 16 and 20 MHz.

hwvern, 2023-11-04:
It only works with a the 20Mhz 328p. I copied the 16Mhz version, and now it also works for this version. Issue #271 is then solved.

MCUdude, 2023-11-04:
> It only works with a the 20Mhz 328p

What only works at 20 MHz? The Urboot bootloader that's being used if you burn the bootloader when the 16 MHz option is selected?

Can you enable _verbose upload_ in the IDE settings and post the output log when you burn the bootloader, and when you try to upload your program using the bootloader?

hwvern, 2023-11-04:
I was wrong; it works with 16Mhz, but not with 20MHz. I burned again with an Nano as ISP, and then uploaded a sketch.
[Burn_bootloader on 16MHz 328p.txt](https://github.com/MCUdude/MiniCore/files/13257688/Burn_bootloader.on.16MHz.328p.txt)
[Burn_bootloader on 20MHz 328p.txt](https://github.com/MCUdude/MiniCore/files/13257689/Burn_bootloader.on.20MHz.328p.txt)
[Upload sketch 16MHz 328p.txt](https://github.com/MCUdude/MiniCore/files/13257690/Upload.sketch.16MHz.328p.txt)
[Upload sketch 20MHz 328p.txt](https://github.com/MCUdude/MiniCore/files/13257691/Upload.sketch.20MHz.328p.txt)

closed 2024-10-10 by MCUdude