MCUdude/MiniCore #273: Upload bootload for external 16mhz don't work with my usbasp programmer

pierrotm777, 2023-11-09:
Hello,

I try to upload your bootloader into my atmega328pb and i can't.
If i upload [this](https://github.com/0x6470/ATmega328PB-Arduino-Nano-Clone-with-Bootloader) i can.
I have just modified the board.txt file for have a 16mhz clock.
Can you solve this issue please.

Thanks,

Pierre

MCUdude, 2023-11-09:
Can you open the IDE settings and turn on verbose upload?
Then post the entire output when you click "burn bootloader", and when you try to upload using the bootloader.

pierrotm777, 2023-11-09:
This is my IDE configuration:
[image]

`avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0\avrdude.conf

         Using Port                    : usb
         Using Programmer              : usbasp
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : usbasp
         Description     : USBasp ISP and TPI programmer
avrdude: auto set sck period (because given equals null)
avrdude usbasp_spi_set_sck_period() error: cannot set sck period; please check for usbasp firmware update
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude usbasp_spi_set_sck_period() error: cannot set sck period; please check for usbasp firmware update

avrdude: processing -U lock:w:0xff:m
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified

avrdude: processing -U efuse:w:0b11110101:m
avrdude: reading input file 0b11110101 for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0b11110101
avrdude: 1 byte of efuse verified

avrdude: processing -U hfuse:w:0xd7:m
avrdude: reading input file 0xd7 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
[... 60 lines cut ...]
avrdude: auto set sck period (because given equals null)
avrdude usbasp_spi_set_sck_period() error: cannot set sck period; please check for usbasp firmware update

avrdude: processing -U flash:w:C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/bootloaders/urboot/atmega328pb/watchdog_1_s/autobaud/uart0_rxd0_txd1/led+b5/urboot_atmega328pb_pr_ee_ce.hex:i
avrdude: reading input file C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/bootloaders/urboot/atmega328pb/watchdog_1_s/autobaud/uart0_rxd0_txd1/led+b5/urboot_atmega328pb_pr_ee_ce.hex for flash
         with 368 bytes in 2 sections within [0x7e80, 0x7fff]
         using 3 pages and 16 pad bytes
avrdude: writing 368 bytes flash ...
Writing | ################################################## | 100% 0.03s
avrdude: 368 bytes of flash written
avrdude: verifying flash memory against C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/bootloaders/urboot/atmega328pb/watchdog_1_s/autobaud/uart0_rxd0_txd1/led+b5/urboot_atmega328pb_pr_ee_ce.hex
Reading | ################################################## | 100% 0.00s
avrdude: 368 bytes of flash verified

avrdude: processing -U lock:w:0xff:m
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified

avrdude done.  Thank you.

`

MCUdude, 2023-11-09:
Can you also print the error you're getting when trying to upload your program?

pierrotm777, 2023-11-09:
I don't understand why but no i can upload the bootloader and upload the blink sketch !
I try no to upload my sketch with a ftdi interface.
This is my schematic. 
[AS5600_FPS148.pdf](https://github.com/MCUdude/MiniCore/files/13311505/AS5600_FPS148.pdf)

MCUdude, 2023-11-09:
It would be very helpful if you did what I asked you to do.

 > Can you also print the error you're getting when trying to upload your program (using the bootloader)?

pierrotm777, 2023-11-09:
Sorry.

`C:\tmp\Arduino_Compilateur\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware -hardware C:\Users\pierrot\AppData\Local\Arduino15\packages -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\tools-builder -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware\tools\avr -tools C:\Users\pierrot\AppData\Local\Arduino15\packages -built-in-libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\libraries -libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries -fqbn=MiniCore:avr:328:bootloader=uart0,eeprom=keep,variant=modelPB,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764 -warnings=all -build-cache C:\Users\pierrot\AppData\Local\Temp\arduino_cache_396638 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Blink\Blink\Blink.ino
C:\tmp\Arduino_Compilateur\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware -hardware C:\Users\pierrot\AppData\Local\Arduino15\packages -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\tools-builder -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware\tools\avr -tools C:\Users\pierrot\AppData\Local\Arduino15\packages -built-in-libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\libraries -libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries -fqbn=MiniCore:avr:328:bootloader=uart0,eeprom=keep,variant=modelPB,BOD=2v7,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764 -warnings=all -build-cache C:\Users\pierrot\AppData\Local\Temp\arduino_cache_396638 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\Blink\Blink\Blink.ino
Using board '328' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0
Detecting libraries used...
"C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\cores\\MCUdude_corefiles" "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\variants\\pb-variant" "C:\\Users\\pierrot\\AppData\\Local\\Temp\\arduino_build_407764\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\cores\\MCUdude_corefiles" "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\variants\\pb-variant" "C:\\Users\\pierrot\\AppData\\Local\\Temp\\arduino_build_407764\\sketch\\Blink.ino.cpp" -o "C:\\Users\\pierrot\\AppData\\Local\\Temp\\arduino_build_407764\\preproc\\ctags_target_for_gcc_minus_e.cpp"
[... cut ...]
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM5

avrdude done.  Thank you.

le port série sélectionné 
 n'existe pas ou votre Arduino n'est pas connectée
`

MCUdude, 2023-11-09:
Can you try running this in cmd.exe? This will execute the same Avrdude upload command (using the bootloader), but will use 250000 baud instead of 115200
```
C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/avrdude.conf -v -patmega328pb -curclock -PCOM5 -b250000 -D -xnometadata -Ueeprom:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.eep:i -Uflash:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.hex:i
```

pierrotm777, 2023-11-09:
Microsoft Windows [version 10.0.22631.2506]
(c) Microsoft Corporation. Tous droits réservés.

C:\Users\pierrot>C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/avrdude.conf -v -patmega328pb -curclock -PCOM5 -b250000 -D -xnometadata -Ueeprom:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.eep:i -Uflash:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.hex:i

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : urclock
         Overriding Baud Rate          : 250000
avrdude: serial_baud_lookup(): using non-standard baud rate: 250000
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM5

avrdude done.  Thank you.

C:\Users\pierrot>
C:\Users\pierrot>As you can see, 115200 baud @ 16 MHz isn't ideal (250000 is), but it's being used for legacy reasons.
'As' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>image
'image' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>@pierrotm777
'pierrotm777' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Add a comment
'Add' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Comment
'Comment' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>
C:\Users\pierrot>Add your comment here...
'Add' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Remember, contributions to this repository should follow our GitHub Community Guidelines.
'Remember' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Assignees
'Assignees' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>No one assigned
'No' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Labels
'Labels' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>None yet
'None' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Projects
'Projects' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>None yet
'None' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Milestone
'Milestone' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>No milestone
'No' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Development
'Development' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>
C:\Users\pierrot>No branches or pull requests
'No' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Notifications
'Notifications' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>Customize
'Customize' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>
C:\Users\pierrot>You’re receiving notifications because you authored the thread.
'You’re' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>2 participants
'2' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>@MCUdude
'MCUdude' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

C:\Users\pierrot>

MCUdude, 2023-11-09:
I just wired up an ATmega328PB, and I can't reproduce the issue. Can you try to replace the 100nF series capacitor on the reset line with a low value resistor (10R) or even better, short it out (0R)? Avrdude 7.2 can actually drive the reset directly, and doesn't really require a capacitor to be there, as long as the USB to serial adapter and the chip runs on the same voltage.

```
Users/hans/Library/Arduino15/packages/MiniCore/tools/avrdude/7.2-arduino.1/bin/avrdude -C/Users/hans/Library/Arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf -v -patmega328pb -curclock -P/dev/cu.usbserial-AH02LQFI -b115200 -D -xnometadata -Ueeprom:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.eep:i -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.hex:i 

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/hans/Library/Arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf
         User configuration file is /Users/hans/.avrduderc

         Using Port                    : /dev/cu.usbserial-AH02LQFI
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Urboot protocol for ATmega328PB
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)

avrdude: processing -U eeprom:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.eep:i
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.eep for eeprom
         with 0 bytes in 0 sections within [0, -1]
         using 0 pages and 0 pad bytes
avrdude: writing 0 bytes eeprom ...
Writing | ################################################## | 100% 0.00s
avrdude: 0 bytes of eeprom written
avrdude: verifying eeprom memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.eep
Reading | ################################################## | 100% 0.00s
avrdude: 0 bytes of eeprom verified

avrdude: processing -U flash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.hex:i
avrdude: reading input file /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.hex for flash
         with 1084 bytes in 1 section within [0, 0x43b]
         using 9 pages and 68 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 1084 bytes flash ...
Writing | ################################################## | 100% 0.19s
avrdude: 1084 bytes of flash written
avrdude: verifying flash memory against /var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_250901/Blink.ino.hex
Reading | ################################################## | 100% 0.24s
avrdude: 1084 bytes of flash verified

avrdude done.  Thank you.

```

pierrotm777, 2023-11-09:
Now, i can upload a bootloader with my usbasp programmer but not with my ftdi interface.

My error:
`C:\tmp\Arduino_Compilateur\arduino-1.8.19\arduino-builder -dump-prefs -logger=machine -hardware C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware -hardware C:\Users\pierrot\AppData\Local\Arduino15\packages -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\tools-builder -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware\tools\avr -tools C:\Users\pierrot\AppData\Local\Arduino15\packages -built-in-libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\libraries -libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries -fqbn=MiniCore:avr:328:bootloader=uart1,eeprom=keep,variant=modelPB,BOD=disabled,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\pierrot\AppData\Local\Temp\arduino_build_697222 -warnings=all -build-cache C:\Users\pierrot\AppData\Local\Temp\arduino_cache_360383 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pierrot\AppData\Local\Arduino15\packages\ATmega328PB-incl-Bootloader\hardware\avr\0.1.0\libraries\Wire\examples\master_reader\master_reader.ino
C:\tmp\Arduino_Compilateur\arduino-1.8.19\arduino-builder -compile -logger=machine -hardware C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware -hardware C:\Users\pierrot\AppData\Local\Arduino15\packages -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\tools-builder -tools C:\tmp\Arduino_Compilateur\arduino-1.8.19\hardware\tools\avr -tools C:\Users\pierrot\AppData\Local\Arduino15\packages -built-in-libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\libraries -libraries C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries -fqbn=MiniCore:avr:328:bootloader=uart1,eeprom=keep,variant=modelPB,BOD=disabled,LTO=Os_flto,clock=16MHz_external -vid-pid=0403_6001 -ide-version=10819 -build-path C:\Users\pierrot\AppData\Local\Temp\arduino_build_697222 -warnings=all -build-cache C:\Users\pierrot\AppData\Local\Temp\arduino_cache_360383 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avrdude.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.avrdude-7.2-arduino.1.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1 -prefs=runtime.tools.arduinoOTA.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\pierrot\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -verbose C:\Users\pierrot\AppData\Local\Arduino15\packages\ATmega328PB-incl-Bootloader\hardware\avr\0.1.0\libraries\Wire\examples\master_reader\master_reader.ino
Using board '328' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0
Detecting libraries used...
"C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\cores\\MCUdude_corefiles" "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\variants\\pb-variant" "C:\\Users\\pierrot\\AppData\\Local\\Temp\\arduino_build_697222\\sketch\\master_reader.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@1.1]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.1]
"C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10819 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\cores\\MCUdude_corefiles" "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\MiniCore\\hardware\\avr\\3.0.0\\variants\\pb-variant" "-IC:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\Min [... cut ...]
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM5

avrdude done.  Thank you.

le port série sélectionné 
 n'existe pas ou votre Arduino n'est pas connectée
`

MCUdude, 2023-11-09:
> avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100

I know FTDI based USB to serial adapters on Windows have been a bit "slow" due to some default driver settings. It may depend from PC to PC.

I've recently replaced Optiboot with Urboot, so a few hiccups are to be expected. Can you try the same command as last time, but this time add `-xstrict` or `xdelay=[delay in milliseconds]`

`
C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/avrdude.conf -v -patmega328pb -curclock -PCOM5 -b250000 -D -xnometadata -Ueeprom:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.eep:i -Uflash:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.hex:i -xdelay=200
`

pierrotm777, 2023-11-09:
Ok, I see that tomorrow morning.
Thanks for your help.
Pierre

pierrotm777, 2023-11-10:
Same issue:
Microsoft Windows [version 10.0.22631.2506]
(c) Microsoft Corporation. Tous droits réservés.

C:\Users\pierrot>C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0/avrdude.conf -v -patmega328pb -curclock -PCOM5 -b250000 -D -xnometadata -Ueeprom:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.eep:i -Uflash:w:C:\Users\pierrot\AppData\Local\Temp\arduino_build_407764/Blink.ino.hex:i -xdelay=200

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : urclock
         Overriding Baud Rate          : 250000
avrdude: serial_baud_lookup(): using non-standard baud rate: 250000
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM5

avrdude done.  Thank you.

MCUdude, 2023-11-10:
Can you try downgrading to the MiniCore v2.2.2 and re-install the bootloader?

pierrotm777, 2023-11-10:
I try Minicore with my IDE v2.2.1.
Bootloader upload is ok.
Sketch return error:
FQBN: MiniCore:avr:328:variant=modelPB
Using board '328' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0
Using core 'MCUdude_corefiles' from platform in folder: C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0

Detecting libraries used...

C:\Users\pierrot\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4/bin/avr-g++ -c -g -Os -std=gnu++17 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_ATmega328 -DARDUINO_ARCH_AVR -Wextra -flto -g -IC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0\cores\MCUdude_corefiles -IC:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.0\variants\pb-variant C:\Users\pierrot\AppData\Local\Temp\arduino\sketches\3328AD8DC4AF82968B8D70091ECE5F39\sketch\Blink.ino.cpp -o nul
exec: "C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist

Compilation error: exec: "C:\\Users\\pierrot\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist

MCUdude, 2023-11-10:
I meant **MiniCore v2.2.2**. The IDE version isn't all that important.

But it looks like you have a broken install of some sort.

> Compilation error: exec: "C:\Users\pierrot\AppData\Local\Arduino15\packages\DxCore\tools\avr-gcc\7.3.0-atmel3.6.1-azduino4/bin/avr-g++": file does not exist

It's trying to use the compiler bundled with DxCore.

You'll have to go into `C:\Users\pierrot\AppData\Local\Arduino15\` and do some cleaning

pierrotm777, 2023-11-10:
It's better , the upload is starting and crashes .

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.2\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 8.0

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: reading input file C:\Users\pierrot\AppData\Local\Temp\arduino_build_438895/Blink.ino.hex for flash
         with 1098 bytes in 1 section within [0, 0x449]
         using 9 pages and 54 pad bytes
avrdude: writing 1098 bytes flash ...

Writing | ######################################------------ | 77% 10.38s

avrdude stk500_recv() error: programmer is not responding

pierrotm777, 2023-11-10:
If i use the programmer STK500 as ISP (Minicore), i can upload with success !!!
[image]

pierrotm777, 2023-11-10:
Just one question, i need to use SDA1/SCL1 as master .
How i can define that ?

pierrotm777, 2023-11-10:
I have again an upload crash , sorry !

avrdude: Version 7.1-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\pierrot\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.2.2\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash                  65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 8.0

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: reading input file C:\Users\pierrot\AppData\Local\Temp\arduino_build_23088/i2c_scanner.ino.hex for flash
         with 4962 bytes in 1 section within [0, 0x1361]
         using 39 pages and 30 pad bytes
avrdude: writing 4962 bytes flash ...

Writing | #####--------------------------------------------- | 10% 10.26s

avrdude stk500_recv() error: programmer is not responding
avrdude stk500_recv() error: programmer is not responding
avrdude stk500_recv() error: programmer is not responding
avrdude stk500_recv() error: programmer is not responding

MCUdude, 2023-11-10:
> I have again an upload crash , sorry !

OK. Then it's very likely hardware-related, as the Optiboot boatload you've now using is very proven and known to be reliable.
One usual suspect is the 16 MHz crystal. The ATmega328PB can't dive cystals and resonators "as hard" as the ATmega328P. Instead it may fail to drive it properly and it can either drift in frequency or stop all together. I'd recommend using a dedicated oscillator that outputs a clock signal that can be fed straight into the XTAL1 pin.

> Just one question, i need to use SDA1/SCL1 as master .

Use the Wire1.h library. `Wire1.begin();`

pierrotm777, 2023-11-10:
Ok, i have tried to define the clock for 8Mhz internal and all is ok.
But it is not enough for my use.
Just for info this bootloader with my 16Mhz quartz is running very well !
https://github.com/0x6470/ATmega328PB-Arduino-Nano-Clone-with-Bootloader/

But it seams that it's only a 8Mhz.
atmega328pb.name=ATmega328PB
atmega328pb.bootloader.file=optiboot_m328pb.hex
atmega328pb.bootloader.unlock_bits=0xFF
atmega328pb.bootloader.lock_bits=0xCF
atmega328pb.bootloader.low_fuses=0xE2
atmega328pb.bootloader.high_fuses=0xDE
atmega328pb.bootloader.extended_fuses=0xFD
atmega328pb.bootloader.tool=avrdude
atmega328pb.upload.tool=avrdude
atmega328pb.upload.protocol=arduino
atmega328pb.upload.maximum_size=32256
atmega328pb.upload.maximum_data_size=2048
atmega328pb.upload.speed=19200
atmega328pb.build.mcu=atmega328pb
atmega328pb.build.board=AVR_UNO
atmega328pb.build.core=arduino:arduino
atmega328pb.build.variant=atmega328pb
atmega328pb.build.f_cpu=8000000L

pierrotm777, 2023-11-10:
I think have solved my upload issue by replacing the avrdude software.
[avrdude.zip](https://github.com/MCUdude/MiniCore/files/13318706/avrdude.zip)

MCUdude, 2023-11-10:
> But it seams that it's only a 8Mhz.

Bingo! The internal 8 MHz is reliable, but "slow" and not as accurate.

Can you try to switch the 22pf capacitors on the XTAL pins with 10pf, 12pf or 15pf?

pierrotm777, 2023-11-10:
Sorry, i have only 22pf capacitors.

pierrotm777, 2023-11-10:
May be can i try to use 2 capa in serial ?

MCUdude, 2023-11-10:
It doesn't hurt to have them in series, but I don't know how reliable it would be...

pierrotm777, 2023-11-10:
I have solved my issue with an external 16 Mhz crystal by using 2 22pf in serial, so 2x11pf is ok.
I use this crystal that need 8pf !
https://fr.rs-online.com/web/p/quartz-electroniques/1446644
Thanks for your help.

closed 2023-11-30 by MCUdude