MCUdude/MiniCore #274: ATMega328PB Analog Comparator Output pin PE0

stevensarns, 2023-11-12:
Thank you for the complete and concise documentation, very clear writing style.  I am unable to get the analog comparator output pin to become active.  I am not sure this is meant to be covered in MilliCore or not.  I am unable to figure out the programming of alternate functions of port E.  I have written the following code running on a 328PB.  The output of the analog comparator is properly reflected in the status of the led and my output pin, but analog comparator output pin remains low (pin 3 of the microcontroller).  Any pointers would be helpful (specifically how to program the alternate function for PE0) - thanks!

`// Board=ATMega328, BOD 2.7, Bootloader yes(com0), clk 20MHz, variant 328PB
// https://github.com/MCUdude/MiniCore#how-to-install
// https://github.com/MCUdude/MiniCore/blob/master/Wiring_reference.md

int  ledPin = PIN_PE2;   // led is PE2 or Arduino 25
bool ACOout;
bool lastACOout;

void setup() {
  Serial.begin(9600);
  pinMode(ledPin, OUTPUT);
  pinMode(PIN_PE3, OUTPUT);
  pinMode(PIN_PE0, OUTPUT); // makes no difference

// ------------------ ACO --------------------------
// ACSR:    ACD   ACBG  ACO ACI ACIE  ACIC  ACIS1 ACIS
// DIDR1:   ---   ---   --- --- ---   ---   AIN1D AIN0D
  ACSR  = B01000000;  // enable=0, AIN+ = 1.1V (BG), no ints
  DIDR1 = B00000011;  // disable digital input buffer
}

void loop() {
  ACOout = bitRead(ACSR,ACO);
  if(ACOout!=lastACOout) {
    if(ACOout) {
      digitalWrite(PIN_PE3,HIGH);
      digitalWrite(ledPin, HIGH);
    }
    else       {
      digitalWrite(PIN_PE3,LOW);
      digitalWrite(ledPin, LOW);
    }
    lastACOout = ACOout;
  }
}`

MCUdude, 2023-11-12:
Hi Steven!

Does the `ACSR` register change when you toggle the comparator input pins?
The code below isn't tested, but it should print the entire status of the ACRS register, and the ACO bit.

```cpp
void setup() {
Serial.begin(9600);

// ------------------ ACO --------------------------
// ACSR: ACD ACBG ACO ACI ACIE ACIC ACIS1 ACIS
// DIDR1: --- --- --- --- --- --- AIN1D AIN0D
ACSR = 0b01000000; // enable=0, AIN+ = 1.1V (BG), no ints
DIDR1 = 0b00000011; // disable digital input buffer
}

void loop() {
  Serial.printf("ACSR reg: 0x%02x, Bit 5: %d\n", ACSR, !!(ACSR & _BV(ACO)));
  delay(500);
}
```

stevensarns, 2023-11-12:
MCUdude, a function generator triangle wave is sent into AIN- pin.  AIN+ pin is 1.1V (Vref).  My code correctly reflects the state of the ACO bit in the ACSR on the led and the digital output PE3 (so yes, ACSR is changing).  But PE0 remains unchanged.

stevensarns, 2023-11-13:
Perhaps my problem is not clear: ACO in ACSR is changing, but the comparator output on pin PE0 is not.

MCUdude, 2023-11-30:
Sorry for not responding earlier. Great to hear you figured it out!

> undocumented register

Damn, that's evil! On the bright side, this issue will probably show up in a Google search if someone is having the same issue with the ATmega324PB/328PB in the future.

closed 2023-11-30 by MCUdude