MCUdude/MiniCore #276: Unable to upload the program to the ATmega328PB at PlatformIO.

kekyo, 2023-11-25:
I use MegaCore and MiniCore. Thanks for the great project.

Please comment if you have any tips as I am unable to upload the program to the ATmega328PB at PlatformIO.

I have designed a board using ATmega328PB and following the Minimal setup schematic in `README.md`.
It runs at 8MHz using the internal clock (with 3.3V VCC). Serial connections are made using CH340N (connected RTS to DTR).
However, LEDs connected to SCK are omitted (D1, R2).

I used Arduino IDE 1.8.19 and added MiniCore 3.0.0 in Board Manager.
Then I connected the AVR ICE and wrote the firmware with the following settings with success.

[image]

In this state, I disconnected the AVR ICE and uploaded a test code like LED Blinker from the COM port to which the CH340N is connected and it worked.

I wanted to try the same configuration with PlatformIO, so I placed the exact same `ino` code in `main.cpp` with the following `platformio.ini`:

```ini
[env:ATmega328PB]
platform = atmelavr
board = ATmega328PB
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 8000000L
```

However, when I try to build and upload the file, nothing to response:

```
Linking .pio\build\ATmega328PB\firmware.elf
Checking size .pio\build\ATmega328PB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.1% (used 206 bytes from 2048 bytes)
Flash: [=         ]   8.0% (used 2596 bytes from 32256 bytes)
Building .pio\build\ATmega328PB\firmware.hex
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Using manually specified: COM5
Uploading .pio\build\ATmega328PB\firmware.hex

(Log stopped)
```

I thought there was a problem with the PlatformIO configuration since it was fine in the Arduino IDE, but am I missing something?

MCUdude, 2023-11-25:
Sorry, but PlatformIO hasn't been updated to use Urboot yet.
I've done all the work, I'll just have to do a bit more testing before [this PR](https://github.com/platformio/platform-atmelavr/pull/321) can merged. Meanwhile, you can burn the bootloader using Arduino IDE, and in platformio.ini, use `upload_protocol = urclock` instead of `arduino`.

kekyo, 2023-11-26:
No problem!  I understood about Urboot after reading the PR and related issues you showed me. Very nice improvements!  Until these are released in the near future, I will continue to work with the Arduino IDE.

Thanks for the clear explanation.

closed 2023-11-26 by kekyo

kekyo, 2023-12-01:
(Sorry I may not be understanding the issue correctly...)

So does that mean that even with the latest version, I can't upload code with PlatformIO?

Maybe if I wrote the bootloader using 3.0.0 in the Arduino IDE (the first way in this issue),
I would have written a bootloader for optiboot version...?
Do I need to write the bootloader for urboot again with PlatformIO with the config value and the following command?

```
board_bootloader.type = urboot
```

```bash
pio run -t bootloader -e fuses_bootloader
```

MCUdude, 2023-12-01:
Sorry, that was a late night typo. It was supposed to be *now* instead of *not* 😅

You can not set the correct fuses, flash the Urboot bootloader and upload using PlatformIO.

kekyo, 2023-12-01:
Working in the middle of the night is what fails :)

Good news, I successfully wrote it in my environment!

1. Connected ATMEL ICE and wrote the bootloader with `pio run -t bootloader -e fuses_bootloader`.
2. after that, I uploaded the code as usual, and it succeeded.

Only one problem remained. That is that the code upload speed only worked at 38400 bps.
However, I assume this was due to the use of the built-in clock generator.
Also, the size of the upload code was small and I did not experience any problem with the write speed for me, so I do not see the need to do anything further.

----

I'll leave the record here in case it might be of help to others.

This is the `platform.ini` at this time:

```ini
[platformio]
default_envs = Upload_UART

[env]
platform = atmelavr  #latest stable
board = ATmega328PB
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 8000000L

[env:Upload_UART]
upload_protocol = urclock
board_upload.speed = 38400

; https://github.com/MCUdude/MiniCore/blob/master/PlatformIO.md
[env:fuses_bootloader]
board_hardware.oscillator = internal
board_bootloader.type = urboot
board_bootloader.speed = 115200
board_hardware.uart = uart0
board_hardware.bod = 2.7v
board_hardware.eesave = yes
upload_protocol = atmelice_isp
```

Log of writing boot loader:

```
C:\Projects\Pixy\Pixy-68000\CoProcessorTests\BlinkerAndSwitch>pio run -t bootloader -e fuses_bootloader
Processing fuses_bootloader (platform: atmelavr; board: ATmega328PB; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/ATmega328PB.html
PLATFORM: Atmel AVR (5.0.0) > ATmega328PB
HARDWARE: ATMEGA328PB 8MHz, 2KB RAM, 32KB Flash
PACKAGES:
 - framework-arduino-avr-minicore @ 3.0.0
 - tool-avrdude @ 1.70200.0 (7.2.0)
 - toolchain-atmelavr @ 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
Building in release mode
Using bootloader image:
C:\Users\k\.platformio\packages\framework-arduino-avr-minicore\bootloaders\urboot\atmega328pb\watchdog_1_s\autobaud\uart0_rxd0_txd1\led+b5\urboot_atmega328pb_pr_ee_ce.hex

TARGET CONFIGURATION:
---------------------
Target = atmega328pb
Clock speed = 8000000L
Oscillator = internal
BOD level = 2.7v
Save EEPROM = yes
Bootloader type = urboot
UART port = uart0
Clock output = no
CFD enable = no
---------------------

Selected fuses: [lfuse = 0xe2, hfuse = 0xd7, efuse = 0xf5]
Setting fuses
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip

avrdude: processing -U lock:w:0xff:m
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified

avrdude: processing -U hfuse:w:0xd7:m
avrdude: reading input file 0xd7 for hfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte hfuse ...
avrdude: 1 byte of hfuse written
avrdude: verifying hfuse memory against 0xd7
avrdude: 1 byte of hfuse verified

avrdude: processing -U lfuse:w:0xe2:m
avrdude: reading input file 0xe2 for lfuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lfuse ...
avrdude: 1 byte of lfuse written
avrdude: verifying lfuse memory against 0xe2
avrdude: 1 byte of lfuse verified

avrdude: processing -U efuse:w:0xf5:m
avrdude: reading input file 0xf5 for efuse
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte efuse ...
avrdude: 1 byte of efuse written
avrdude: verifying efuse memory against 0xf5
avrdude: 1 byte of efuse verified

avrdude done.  Thank you.

Uploading bootloader
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip

avrdude: processing -U flash:w:C:\Users\k\.platformio\packages\framework-arduino-avr-minicore\bootloaders\urboot\atmega328pb\watchdog_1_s\autobaud\uart0_rxd0_txd1\led+b5\urboot_atmega328pb_pr_ee_ce.hex:i
avrdude: reading input file C:\Users\k\.platformio\packages\framework-arduino-avr-minicore\bootloaders\urboot\atmega328pb\watchdog_1_s\autobaud\uart0_rxd0_txd1\led+b5\urboot_atmega328pb_pr_ee_ce.hex for flash
         with 368 bytes in 2 sections within [0x7e80, 0x7fff]
         using 3 pages and 16 pad bytes
avrdude: writing 368 bytes flash ...
Writing | ################################################## | 100% 0.04s
avrdude: 368 bytes of flash written
avrdude: verifying flash memory against C:\Users\k\.platformio\packages\framework-arduino-avr-minicore\bootloaders\urboot\atmega328pb\watchdog_1_s\autobaud\uart0_rxd0_txd1\led+b5\urboot_atmega328pb_pr_ee_ce.hex
Reading | ################################################## | 100% 0.00s
avrdude: 368 bytes of flash verified

avrdude: processing -U lock:w:0xFF:m
avrdude: reading input file 0xFF for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xFF
avrdude: 1 byte of lock verified

avrdude done.  Thank you.

=============================================== [SUCCESS] Took 1.36 seconds ===============================================

Environment       Status    Duration
----------------  --------  ------------
fuses_bootloader  SUCCESS   00:00:01.356
=============================================== 1 succeeded in 00:00:01.356 ===============================================
```