MCUdude/MiniCore #277: LTO problems

Floessie, 2023-11-26:
Hi Hans,

I just wanted to let you know that there seems to be a problem with LTO enabled code. My current setup is Arduino IDE 1.8.19 on Debian AMD64 with the latest MiniCore 3.0.0. The board is a [custom PCB](https://github.com/Floessie/CavyCave-hardware) with a 328p at 8MHz external (but I also had LTO problems in another project with 4MHz internal). I've also published the firmware for the board [here](https://github.com/Floessie/CavyCave-firmware), if you are interested, but the problem manifests itself in a very simple library.

If LTO is enabled, the used TroykaDHT library is unable to sample the DHT22 sensor. This can be easily reproduced with the [TroykaDHT example file](https://github.com/amperka/TroykaDHT/blob/master/examples/testDHT/testDHT.ino). If LTO is disabled, everything works as expected.

I do remember that my projects worked well with LTO enabled in the past, so either the problems originate in the Arduino IDE updates over the years or in a toolchain update in MiniCore.

Maybe LTO should not be made available as long as there are problems (obviously with `pulseInLength()` timing as used in TroykaDHT).

Best,
Flössie

MCUdude, 2023-11-30:
I'm using the same compiler as the official Arduino core does. MiniCore has been using the same compiler for years, so the only thing that has changed is that LTO is now enabled by default. The official Arduino core also enables LTO by default and doesn't let the user disable it. v3.0.0 is a major revision bump. Don't be too surprised if there are breaking changes.

Floessie, 2023-12-06:
Hi Hans,

As said, I just wanted to let you know that something is wrong with LTO. It took me some time to figure out, so I hope others can benefit from my findings: If something is odd (especially with timing code), first disable LTO.

HTH,
Flössie

closed 2023-12-06 by Floessie