MCUdude/MiniCore #279: ATmega328PB-AU - signature mismatch

fenrir-5, 2023-11-29:
Hello all,
I'm trying to burn the Arduino bootloader to the ATmega328PB-AU TQFP32 MCU using Arduino as ISP and I get all kinds of avrdude errors. For some reason it worked on few chips but I couldn't upload my sketch via FTDI using the Sparkfun FTDI basic UART board.
Here's what I get in the console when I try to burn the bootloader:
`Arduino: 1.8.19 (Linux), Board: "ATmega328, Yes (UART0), EEPROM retained, 328PB, BOD 2.7V, LTO disabled, External 16 MHz"

/home/fenrir/.arduino15/packages/MiniCore/tools/avrdude/7.2-arduino.1/bin/avrdude -C/home/fenrir/.arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf -v -patmega328pb -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0b11110101:m -Uhfuse:w:0xd7:m -Ulfuse:w:0b11111111:m 

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/fenrir/.arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf
         User configuration file is /home/fenrir/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x0000ff
avrdude main() error: expected signature for ATmega328PB is 1E 95 16
        double check chip or use -F to override this check

avrdude done.  Thank you.

Error while burning bootloader.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
Any idea how can I make it work?
Thank you!

MCUdude, 2023-11-29:
I wouldn't be surprised if this was caused by the external 16 MHz clock. The ATmega328PB is very delicate, and can't drive crystals/load capacitor combinations that aren't perfectly matched, unlike the ATmega328P, which could drive pretty much anything. How many pF are your load capacitors on the crystal?

What happens if you remove the crystal and load capacitors, use [this Arduino as ISP sketch](https://github.com/adafruit/ArduinoISP) instead, and connect pin 9 from the Arduino as ISP board to the XTAL1 pin on the 328PB? This will ensure you have a proper, driven clock. If programming succeeds, it means that the 328PB couldn't drive the crystal properly. I've used the _very_ similar ATmega324PB in a commercial design where close to a thousand units were made, and crystals that stopped oscillating were a big problem. I ended up replacing the crystal with a proper external clock source that fed a square wave straight into the XTAL1 pin. It only cost a few cents more than a crystal, but was well worth it.

fenrir-5, 2023-11-29:
Thanks a lot for the quick reply, MCUdude.
> The ATmega328PB is very delicate
That's great, we're f'ed as the deadline very close and we have to assemble and ship 15 working boards until the end of this week.
I didn't remove the crystal but what I tried was to place it closer to the MCU by soldering its pins on the test socket so potential issues like the length of wires or breadboard capacitance are eliminated.
Regarding capacitors, I used 22pF as I did before when worked with the 328P variant.
I will desolder them in a few minutes and give it another try using  your ISP sketch.
Thanks!
[image]

MCUdude, 2023-11-29:
You _could_ try to replace the 22pf capacitors with some smaller ones, like 15 or even 10 pF. If you don't have that, try putting two in series. That will half their value

fenrir-5, 2023-11-30:
Ok, this getting weird. I desoldered the crystal and 22pF capacitors from the test socket pins and I tried to burn the bootloader with the internal 8 MHz selected. Nothing good in the console.
Then I tried with the external 16 MHz crystal and 11 pF capacitors (22 pF in series as you recommended) and still nothing new in the console.
Then I opened the test socket, changed the chip with a new one and got this:
`
Arduino: 1.8.19 (Linux), Board: "ATmega328, Yes (UART0), EEPROM retained, 328PB, BOD  disabled, LTO disabled, External 16 MHz"

/home/fenrir/.arduino15/packages/MiniCore/tools/avrdude/7.2-arduino.1/bin/avrdude -C/home/fenrir/.arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf -v -patmega328pb -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0b11110111:m -Uhfuse:w:0xd7:m -Ulfuse:w:0b11111111:m 

avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /home/fenrir/.arduino15/packages/MiniCore/hardware/avr/3.0.0/avrdude.conf
         User configuration file is /home/fenrir/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/ttyACM0
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e9516 (probably m328pb)
avrdude: erasing chip

avrdude: processing -U lock:w:0xff:m
avrdude: reading input file 0xff for lock
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte lock ...
avrdude: 1 byte of lock written
avrdude: verifying lock memory against 0xff
avrdude: 1 byte of lock verified

avrdude: processing -U efuse:w:0b11110111:m
avrdude: reading input file 0b11110111 for efuse
[... 83 lines cut ...]
         using 3 pages and 16 pad bytes
avrdude: writing 368 bytes flash ...
Writing | ################################################## | 100% 0.09s
avrdude: 368 bytes of flash written
avrdude: verifying flash memory against /home/fenrir/.arduino15/packages/MiniCore/hardware/avr/3.0.0/bootloaders/urboot/atmega328pb/watchdog_1_s/autobaud/uart0_rxd0_txd1/led+b5/urboot_atmega328pb_pr_ee_ce.hex
Reading | ################################################## | 100% 0.00s
avrdude avr_verify() warning: verification mismatch
        device 0x40 != input 0x11 at addr 0x7e80 (error)
        device 0x40 != input 0x24 at addr 0x7e81 (error)
        device 0x41 != input 0x24 at addr 0x7e82 (error)
        device 0x41 != input 0xb6 at addr 0x7e83 (error)
        device 0x42 != input 0x14 at addr 0x7e84 (error)
        device 0x42 != input 0xbe at addr 0x7e85 (error)
        device 0x43 != input 0x80 at addr 0x7e86 (error)
        device 0x43 != input 0xe0 at addr 0x7e87 (error)
        device 0x44 != input 0x7e at addr 0x7e88 (error)
        device 0x44 != input 0xd0 at addr 0x7e89 (error)
        suppressing further verification errors
avrdude do_op() error: verification mismatch

avrdude done.  Thank you.

Error while burning bootloader.
`
I'm already starting to lose hope that I can get something good using this variant of 328 and the only option is to buy 328P-AU from a local store asap.

MCUdude, 2023-11-30:
The ATmega328PB isn't useless, but its clock circuit is delicate and finicky and requires attention when designing it. If matching the crystal/capacitors or using a dedicated external clock isn't an option, use the 328P instead.

closed 2024-05-23 by MCUdude