MCUdude/MiniCore #280: uP_table does not know mcuid 1753

miclol, 2023-12-10:
I'm trying to upload a sketch to an ATMega328P via an Arduino UNO, but it throws an error. I am using V3.0.0 of Minicore.
Traceback:
```
Sketch uses 3976 bytes (12%) of program storage space. Maximum is 32384 bytes.
Global variables use 195 bytes (9%) of dynamic memory, leaving 1853 bytes for local variables. Maximum is 2048 bytes.
avrdude warning: uP_table does not know mcuid 1753
avrdude warning: programmer is not responding
avrdude error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
the selected serial port         - use -F to override this check
 does not exist or your board is not connected
```

miclol, 2023-12-10:
After some testing, this seems to be caused by the bootloader, as disabling it fixes the issue.

MCUdude, 2023-12-10:
It may be because the internal oscillator isn't all that accurate, and there is a slight baud rate error that causes Avrdude to read an incorrect mcuid. Did you just select the ATmega328P, internal 8 MHz clock, and then hit "Burn bootloader"?

miclol, 2023-12-10:
Yes, burning the bootloader causes no problems. Only when uploading sketches does the error occur.

MCUdude, 2023-12-10:
I'll have to ask @stefanrueger who wrote the bootloader + the Avrdude implementation for help. It might be because of the serial connection, but let's hear what he has to say.

stefanrueger, 2023-12-10:
If it is an autobaud bootloader using a different host baud rate might do the trick. In absence of more debugging info (avrdude version, command line, baud rate, which urboot bootloader configuration, the *actual* frequency of the internal oscillator, ...), @MCUdude's [conjecture](https://github.com/MCUdude/MiniCore/issues/280#issuecomment-1848879805) is the best I can come up with as well.

> upload a sketch to an ATMega328P via an Arduino UNO

I don't understand the "via an Arduino UNO" bit. If the target ATmega328P wants to use its urboot bootloader then the target's serial lines (D0/D1) are supposed to be connected directly to the host PC, possibly via a USB to serial connector cable. How is the connection rigged up?

MCUdude, 2023-12-10:
The problem with Arduino IDE is that it's not all that easy to let the user specify the upload speed, as it would also have to match the main clock to be withing the acceptable ±2.5% error margin.

What I _could_ do was to create a new menu option, "Upload speed", where I provide a set of common baud rates, but also a "default" one that sits on the very top of the list. Then the user could use the default one in most cases, but at the same time have the freedom to either increase or descrease the speed is needed. Sounds like a feature for the 3.0.1 release!

closed 2024-07-26 by MCUdude