MCUdude/MiniCore #281: Upload error: cannot set com-state on COM port and unable to open programmer error

MewnianSquire, 2024-01-22:
```
avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Qbar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1\avrdude.conf

         Using Port                    : COM3
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
avrdude ser_open() error: cannot set com-state for \\.\COM3
avrdude main() error: unable to open programmer urclock on port COM3

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected

```

I uploaded a new bootloader into a cheap nano v3 clone that uses a 328PB using a genuine UNO R3 that uses a 328P. I connected the nano like the one in [this](https://support.arduino.cc/hc/en-us/articles/4841602539164-Burn-the-bootloader-on-UNO-Mega-and-classic-Nano-using-another-Arduino) guide from arduino and followed the steps on the readme, the burn was successful. But when I tried to upload blink example it didn't work and came back with the error above. Oddly I can send serial data through the serial port and the Rx led blinks. I haven't found any solutions yet, so any feedback is appreciated

raphtronicsTech, 2024-01-22:
I had the same issue although with ATMEGA328P - PU chip. I was able to resolve the issue by downgrading the board's version by installing an old version of the board.

MewnianSquire, 2024-01-23:
> by installing an old version of the board.

How does one do that?

MCUdude, 2024-01-23:
You need to downgrade the CH340 drivers to an earlier version. There are lots of similar cases on the Arduino forum.

raphtronicsTech, 2024-01-23:
> How does one do that?

Go to your board manager in the Arduino IDE, click on the version drop down list and select an older version of the minicore board then click on install.

MCUdude, 2024-01-23:
> Go to your board manager in the Arduino IDE, click on the version drop down list and select an older version of the minicore board then click on install.

How would this work? The issue he's facing is most likely related to the latest WCH CH340 USB to serial driver on Windows. It has been commonly reported lately, and there is a fix for it:

Read this:
https://forum.arduino.cc/t/avrdude-ser-open-cant-set-com-state-for-com3/1209419

> I had the same issue although with ATMEGA328P - PU chip. I was able to resolve the issue by downgrading the board's version by installing an old version of the board.

If this is the case, I'd like to hear about it, ideally in a separate issue. The latest version of MiniCore should always be "better" than the previous one and be just as compatible.

MewnianSquire, 2024-01-24:
Okay an update:
1. I followed the CH340 driver rollback steps with little luck, the IDE can open the port now but still can't upload.
2. Tried the old bootloader and minicore bootloader with no luck.

Error messages:
Minicore bootloader:
```
Arduino: 1.8.19 (Windows 10), Board: "ATmega328, Yes (UART1 328PB only), EEPROM retained, Default, 328PB, BOD 2.7V, LTO enabled, External 16 MHz"

avrdude: Version 7.2-arduino.1

         Copyright the AVRDUDE authors;

         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Qbar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1\avrdude.conf

         Using Port                    : COM3

         Using Programmer              : urclock

         Overriding Baud Rate          : 115200

avrdude urclock_getsync() warning: attempt 1 of 10: not in sync

avrdude urclock_getsync() warning: attempt 2 of 10: not in sync

avrdude urclock_getsync() warning: attempt 3 of 10: not in sync

avrdude urclock_getsync() warning: attempt 4 of 10: not in sync

avrdude urclock_getsync() warning: attempt 5 of 10: not in sync

avrdude urclock_getsync() warning: attempt 6 of 10: not in sync

avrdude urclock_getsync() warning: attempt 7 of 10: not in sync

avrdude urclock_getsync() warning: attempt 8 of 10: not in sync

avrdude urclock_getsync() warning: attempt 9 of 10: not in sync

avrdude urclock_getsync() warning: attempt 10 of 10: not in sync

avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100

avrdude main() error: unable to open programmer urclock on port COM3

avrdude done.  Thank you.

the selected serial port 

 does not exist or your board is not connected

```

Old bootloader error:
```
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Qbar\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0xc9
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xc9

avrdude done.  Thank you.

```

I do really think this is an issue with the CH340 on the board... well I don't know if it's actually a CH340 since the IC has no markings, but it is shaped like the CH340 on my esp32 so...

MCUdude, 2024-01-24:
If you're downgrading to MiniCore v2.x, you'll have to burn the bootloader again, and v2.x uses a different bootloader than v3.x

I would be skeptical if the chip had no markings. It may be a rejected CH340 chip that someone picked up and sold on the grey market.

But regarding the error message you posted, this looks like progress. Now Avrdude can at least open the serial port:

```
Error messages:
Minicore bootloader:

Arduino: 1.8.19 (Windows 10), Board: "ATmega328, Yes (UART1 328PB only), EEPROM retained, Default, 328PB, BOD 2.7V, LTO enabled, External 16 MHz"

avrdude: Version 7.2-arduino.1

         Copyright the AVRDUDE authors;

         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Qbar\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1\avrdude.conf

         Using Port                    : COM3

         Using Programmer              : urclock

         Overriding Baud Rate          : 115200

avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
 [...]
```

However, it can't communicate with the chip. It may be the auto-reset circuit, which is difficult to verify without an oscilloscope, but it can also be the USB to serial chip. Do the RX/TX LEDs flash when you try to upload? And to you have a link to where you ordered the board from?

MewnianSquire, 2024-01-24:
> Do the RX/TX LEDs flash when you try to upload?
No. But when I press send on the serial monitor on the old IDE the Rx led flashes

>  And to you have a link to where you ordered the board from?
It's an Indonesian website called [Tokopedia.](https://www.tokopedia.com/hwthinker/nano-3-0-type-c-typec-atmega328p-atmega-328p-board-for-arduino-ide-kabel-usb-typec?utm_campaign=PDP-22303120-5745285594-0-240124-default&utm_source=salinlink&utm_medium=share). It's in Indonesian but it has many reviews already and the seller is reputable

MCUdude, 2024-01-24:
> Arduino: 1.8.19 (Windows 10), Board: "ATmega328, Yes (UART1 328PB only)

But have you flashed the correct bootloader? From what I can tell, you're using UART1 for uploading, but the Arduino NANO form factor (and the ATmega328P) uses UART0

MewnianSquire, 2024-01-24:
> But have you flashed the correct bootloader? From what I can tell, you're using UART1 for uploading, but the Arduino NANO form factor (and the ATmega328P) uses UART0

I think so... the chip on the board itself is marked 328PB so I used UART1. I'll try using the using the bootloader with UART0 later since I am not able to do that right now....

harryberlin, 2024-01-24:
for UART1 you need to use the other Pins.

MewnianSquire, 2024-01-27:
I've managed to get the nano to work using UART0 and an older version of the CH driver

closed 2024-01-27 by MewnianSquire

MCUdude, 2024-01-27:
Excellent! Thanks for reporting back