MCUdude/MiniCore #282: Ch340 upload problem

ecvary, 2024-01-31:
I tried uploading the ATMega328 code using ch340, and the error message occurred. 
``
Arduino: 1.8.19 (Windows 7), Board: "ATmega328, Yes (UART0), EEPROM retained, Default, 328P / 328PA, BOD 2.7V, LTO enabled, External 16 MHz"

Sketch uses 880 bytes (2%) of program storage space. Maximum is 32384 bytes.

Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
avrdude warning: attempt 1 of 10: not in sync
avrdude warning: attempt 2 of 10: not in sync
avrdude warning: attempt 3 of 10: not in sync
avrdude warning: attempt 4 of 10: not in sync
avrdude warning: attempt 5 of 10: not in sync
avrdude warning: attempt 6 of 10: not in sync
avrdude warning: attempt 7 of 10: not in sync
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open programmer urclock on port COM7
avrdude error: unable to open programmer urclock on port COM7
``
but if you use usbasp, the program can be uploaded and no problems will occur. I have tried using 2 ch340 programmers and 2 minimum system board , no change. 
the driver works, and the programmer is detected in device manager, all cables are connected.
how to  fix it?. Thank You

MCUdude, 2024-01-31:
What does your hardware look like? Does the board have a functioning auto-reset circuitry?

ecvary, 2024-01-31:
Yes,  
 DTR to reset pin atmega328.
RX to TX atmega328
TX to RX  atmega328

closed 2024-04-15 by ecvary