MCUdude/MiniCore #286: avrdude urclock_res_check() error:

briansbrian, 2024-02-14:
**`
         Using Port                    : /dev/ttyCH341USB0
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Bootloader using STK500v1 communication protocol
avrdude urclock_res_check() error: protocol expects sync byte 0x14 but got 0x00 in urclock_program_enable()
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude urclock_disable() error: protocol expects sync byte 0x14 but got 0x10

avrdude done.  Thank you.

the selected serial port 
 does not exist or your board is not connected**
`

MCUdude, 2024-02-14:
What does your hardware look like? Are you using the internal oscillator? And which clock speed?

briansbrian, 2024-02-14:
[image]
external crystal oscilator- 16Mhz

briansbrian, 2024-02-14:
# This is a simple code i tried after countless hours of trying to upload examples which failed...

 
**int Led = 7;
void setup() {
  int Led = 7;
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(Led , OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(Led , HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(Led , LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}**

## IT FAILED!  ....BUT I ADDED <sup>-xbootsize=8192 </sup> in the  cmd
###THIS IS WAS THE OUTPUT
**avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\lab\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1\avrdude.conf

         Using Port                    : COM3
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328PB
         Chip Erase delay              : 10500 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                 65    20     4    0 no       1024    4      0  3600  3600 0x00 0x00
           flash                  65    10   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Bootloader using STK500v1 communication protocol
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x62950f (probably m328pb)

avrdude: processing -U eeprom:w:C:\Users\lab\AppData\Local\Temp\arduino_build_45417/Blink.ino.eep:i
avrdude: reading input file C:\Users\lab\AppData\Local\Temp\arduino_build_45417/Blink.ino.eep for eeprom
         with 0 bytes in 0 sections within [0, -1]
         using 0 pages and 0 pad bytes
avrdude: writing 0 bytes eeprom ...
Writing | ################################################## | 100% 0.00 s
avrdude: 0 bytes of eeprom written

avrdude: processing -U flash:w:C:\Users\lab\AppData\Local\Temp\arduino_build_45417/Blink.ino.hex:i
avrdude: reading input file C:\Users\lab\AppData\Local\Temp\arduino_build_45417/Blink.ino.hex for flash
         with 1098 bytes in 1 section within [0, 0x449]
         using 9 pages and 54 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 1098 bytes flash ...
Writing | ################################################## | 100% 0.19 s
avrdude: 1098 bytes of flash written

avrdude done.  Thank you.**

MCUdude, 2024-02-14:
It looks like the chip already has Optiboot installed:

`Bootloader using STK500v1 communication protocol`

Please re-burn the bootloader using MiniCore. That will install Urboot, which is the bootloader MiniCore now uses.

briansbrian, 2024-02-14:
**I cant understand the read.md
Please can you detail me the steps**

MCUdude, 2024-02-14:
I'm not going to hold your hand through this. MiniCore is a 3rd party addon to Arduino IDE to provide more advanced functionality.

Follow this guide or search online if you need help burning the bootloader:
https://github.com/MCUdude/MiniCore?tab=readme-ov-file#getting-started-with-minicore

ecvary, 2024-02-15:
It seems that Urboot can only be programmed using ISP.

MCUdude, 2024-02-15:
> It seems that Urboot can only be programmed using ISP.

Correct.

briansbrian, 2024-02-15:
thanks guys

closed 2024-03-14 by MCUdude